uweseimet/scsi2pi #8: Test SASI hard drive support

uweseimet, 2023-12-15:
After re-adding SASI hard drive support this has so far only been tested with an Atari and a tweaked Atari SCSI driver version. Since the Atari usually speaks SCSI these tests only have a limited value.
Help with testing with a more realistic SASI setup is appreciated.

uweseimet, 2023-12-15:
@hlide @wohali In https://github.com/PiSCSI/piscsi/issues/809 you expressed interest in SASI being supported again for the PiSCSI/RaSCSI port. In the new SCSI2Pi project I re-added SASI support and cleaned up the respective codebase.
I would appreciate your help with testing.

uweseimet, 2023-12-25:
@wohali Good news, thank you for your feedback! Please get back to me when you are ready.

wohali, 2024-01-31:
Hi @uweseimet , I literally just got my replacement PiSCSI HW yesterday. Let me get it together and try and run a vintage SASI system. It'll be no sooner than this weekend.

philpem, 2024-02-01:
I need to have a look at the setup but I have a colleague's Acorn Winchester drive on the bench (bad Winchester disk) so may be able to do a quick test. Same problem as Wohali though, I won't be able to do it until the weekend.

That'd be the Winchester drive and host adapter in an Acorn BBC Master 128. I think the standard controller is an Adaptec ACB, possibly an ACB4000

uweseimet, 2024-02-01:
@philpem Sounds good. Please use the current SCSI2Pi 2.0 release or the develop branch for testing.

wohali, 2024-02-05:
hi @uweseimet - just letting you know this is not forgotten, physically moving the system to somewhere i can test it easily requires assistance and that didn't happen this weekend. I'm hoping early this week.

I have my PiSCSI assembled and will use the `develop` branch for testing.

uweseimet, 2024-03-03:
Looks is if there is no serious interest in SASI support. *Serious* interest means that somebody tests the existing code and reports issues until they are fixed.
Note that if in the future nobody is seriously testing SASI I am going to remove SASI support for good.

wohali, 2024-03-03:
@uweseimet Uwe, I appreciate all your hard work.

For me, real life must come first. I am quite old, have had illnesses, and can no longer carry heavy equipment. The individual I need to help me carry things has not been available for weeks now, due to situations out of their control. I, too, have been dealing with serious illness that required hospitalisation - but that is not within the scope of this request.

Do what you must with the code, but please, do not get angry with me. I feel your frustration, but your anger is misplaced. My ability to test this functionality largely remains out of my control as well.

eschaton, 2024-03-03:
There was more than Atari that used SASI, it was popular since its introduction for 68000-based minicomputer systems, as well as the Apple II "Sider" hard disk system as well as other Xebec stuff. I think the X68000 and PC-98 also used it. It'd be worthwhile to ping those communities to see whether anyone can help test. (I should have at least one 68000-based mini that uses SASI.)

philpem, 2024-03-04:
Uwe, I understand your frustrations, but I have similar issues to Wohali. A busy day-job, and a lack of free space due to an ongoing spare-time task.

The Winchester I have here is the later version with an ACB4000 bridge board, not the earlier Xebec. The Adaptec documentation calls it SCSI. The Acorn documentation uses both terms. So I'm not sure whether it would truly be a SASI test.

The only other machine I have which may be SASI is an ACT/Sirius (Victor 9000) with the hard drive option, but it needs the power supply and floppy drives repairing before I can use it to test SASI.

Some helpful documents I found while I was just looking this up...
SASI spec: http://bitsavers.trailing-edge.com/pdf/shugart/SASI/SASI_Design_Specifications_Rev_F_198210.pdf
ACB4000 manual: http://bitsavers.trailing-edge.com/pdf/adaptec/ACB-4000/ACB_4000_and_5000_Disk_Controllers_OEM_Manual_Preliminary_May83.pdf

uweseimet, 2024-03-04:
I am not frustrated or angry, but it's a sad that once again, just like with PiSCSI, interest in SASI is claimed, but effectively nobody is testing/using it. Before being dropped from PiSCSI about 1 1/2 years ago, SASI had not been working anymore for several months (it immediately crashed) but nobody noticed. When you look at the SASI-related PiSCSI tickets, nobody appears to have been using SASI with PiSCSI for at least 2 years.
Nevertheless I added SASI support to SCSI2Pi, also as a proof of concept. I contacted all users (I think) who had expressed their interest in the PiSCSI SASI reactivation. But history seems to repeat itself: Interest in SASI is claimed but nobody (for various reasons) uses it.

I am not going to remove SASI support *today*, but if there is nobody using/testing SASI in the next couple of months it does not make sense to keep the code in the repository. Any code that is not used means more maintainance work, makes the binaries bigger etc. And it is no fun to deal with software that is not being used ;-).

Regarding SASI and the Atari: SASI is not relevant for Atari computers. Usually SCSI, IDE/SATA and nowadays even USB drives are used with these machines. The first hard drives offered by Atari for the ST series (SH and Megafile drives) used Adaptec controllers. But soon adapters for SCSI drives became state of the art. Atari themselves started using SCSI drives with the MegaSTE and TT. The TT and Falcon030 have fully-featured SCSI hardware based on the NCR 5380 SCSI controller.

uweseimet, 2024-03-04:
@philpem Thank you for your document links. Some information was new to me, and most likely it will help me to merge some of the SASI and SCSI controller code, which will reduce the overhead of supporting SASI.

hlide, 2024-10-11:
Ok, I have an X68000 PRO II with PhantomX to help me to configure it. have only the original RaSCSI board but I guess it is supposed to work with PiSCSI firmware, right? I must check whether the SCSI cable I have for a real SCSI CDROM device is fully compatible. And check how to install all that stuff.

uweseimet, 2024-10-11:
@hlide Regarding SCSI2Pi, you either need a STANDARD or a FULLSPEC board. Other board flavors are not supported anymore.

hlide, 2024-10-11:
The cable is not the right one :(. That hell of SCSI cable jungle... I must buy one.

uweseimet, 2024-11-28:
I does not make sense to keep this ticket open any longer. I am not aware of bugs, nobody has reported a bug, let's just asume everything is fine ;-).

closed 2024-11-28 by uweseimet