uweseimet/scsi2pi #23: Support BlueSCSI compatibility mode for names of drive image files

uweseimet, 2024-01-16:
BlueSCSI codes drive properties in the name of the image files. While the command line approach of SCSI2Pi/PiSCSI is more flexible it would be useful to have a compatibility mode with the same parsing rules as those used by BlueSCSI, see https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Usage.
A new option "-B" shall be added to s2p. With this option s2p takes the drive parameters from the filename.
Instead of
```
>s2p -i 0:2 -b 1024 -n "vendor:product:revision test.hds
```
you can use
```
>s2p -B HD02_1024vendor:product:revision.hds
```
The existing command line options shall override the settings taken from the filename:
```
>s2p -B HD10_1024.hds -i 2
```
This command assigns the drive a SCSI ID of 2, because "-i" overrides the ID coded in the filename.

uweseimet, 2024-01-16:
@erichelgeson I would appreciate your feedback on this ticket. The proposed parsing rules would also make use of the LUN field, which appears to be unused in BlueSCSI. They also use the unused (in BlueSCSI) text between the sector size and the extension as the INQUIRY product data.
Would you expect potential compatibility issues with future BlueSCSI versions?

uweseimet, 2024-01-16:
The issue_23 branch now contains unit-tested support for this compatibility mode.

uweseimet, 2024-01-17:
Merged into develop branch

closed 2024-01-17 by uweseimet

erichelgeson, 2024-01-28:
My 2c adding vendor/product/version is not needed. The "presets" we have defined would be a better option as there are usually other things that need to be set to be compatible with different host systems .

uweseimet, 2024-01-28:
@erichelgeson Thank you for your feedback. As far as the SCSI2Pi compatibility mode is concerned only the filenames matter for this mode. This is not about further compatibility. Any other BlueSCSI configuration mechanism would be out of scope for SCSI2PI, which has its own configuration mechanism for defining custom mode pages etc.
The option of having the product data in the filename would just be something I think is convenient, because SCSI ID/LUN, sector size and product name are the most frequent configuration properties as far as I can tell.