uweseimet/scsi2pi #33: Documentation should mention that libprotobuf32 package needs to be installed to use s2p

fdanapfel, 2024-01-20:
# When creating a new ticket please provide information on your environment.

- The Pi type: 
Zero 2 W

- The Pi OS version (output of 'lsb_release -a'):
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm

- The SCSI2Pi release or git revision: 
scsi2pi_1.2_devel_fb4e56b_arm64-1.deb

- The computer/sampler connected to the RaSCSI/PiSCSI board:
N/A

# Describe the issue
When trying to start s2p on a fresh installation of Raspberry PI OS 12 (bookworm) arm64 lite after installing the scsi2p .deb as documented on https://www.scsi2pi.net/en/downloads.html, starting s2p fails with the following error:

$ sudo /opt/scsi2pi/bin/s2p -id 6 daynaport
./s2p: error while loading shared libraries: libprotobuf.so.32: cannot open shared object file: No such file or directory

To fix this libprotobuf32 needs to be installed:
$ sudo apt install libprotobuf32

Afterwards s2p starts without error.

It would therefore be good if either a dependency to libprotobuf32 could be added to the .deb packages, or if it was mentioned somewhere on https://www.scsi2pi.net/en/downloads.html that this package must be installed when using a fresh install of Raspberry Pi OS.

It might also be good to mention that s2p needs to be started using sudo, since it requires root privileges. Trying to start s2p as a user with admin privileges fails because GPIO access isn't allowed.

uweseimet, 2024-01-20:
@fdanapfel Thank you for your feedback. I agree regarding the dependencies. Currently I still assume that piscsi is already installed, which ensures that all the dependencies are met. But I am already planning to add dependency information to the installation package.
Regarding s2p and sudo, this is the same as with piscsi, which also requires root permissions. I will update the website with information on this. I guess there may also still be other things that should be added to the website, but this is just a matter of time.

uweseimet, 2024-01-20:
The website now uses sudo in the sample commands.

uweseimet, 2024-01-20:
I added a dependency to 'libprotobuf-dev' to the installation rules. This also installs the header files, which are not needed in the binary package, but libprotobuf32 is OS-dependant. The library version depends on the protobuf compiler version used. For bullseye it is libprotobuf23. 'libprotobuf-dev' appears to be the only package name that works with both bullseye and bookworm. (It also worked with buster.)
If everything goes well the next nightly builds (tomorrow very early morning) will include this change.

uweseimet, 2024-01-21:
A new binary package is available. dpkg now reports the missing dependency. It cannot automatically install libprotobuf-dev, though, because apt is needed for that. https://www.scsi2pi.net/en/downloads.html has the details.

It is probably acceptable to set ui root for s2p, s2pdump and s2pexec during the installation, because usually you run them with root permissions on the Pi anyway. Another develop branch build for testing this is already running.

fdanapfel, 2024-01-21:
@uweseimet Thanks for taking a look at this and providing a fix so quickly.

BTW. it is also possible to use apt to install a local .deb file, which will automatically resolve and install the dependencies:
```
$ sudo apt install ./scsi2pi_1.2_devel_50d3589_arm64-1.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'scsi2pi' instead of './scsi2pi_1.2_devel_50d3589_arm64-1.deb'
The following additional packages will be installed:
  libprotobuf-dev libprotobuf-lite32 libprotobuf32 zlib1g-dev
The following NEW packages will be installed:
  libprotobuf-dev libprotobuf-lite32 libprotobuf32 scsi2pi zlib1g-dev
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 3,193 kB/3,868 kB of archives.
After this operation, 17.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
```

uweseimet, 2024-01-21:
@fdanapfel I have already tried using apt this way, but for me it failed:
```
>sudo apt install ./scsi2pi_.2_devel_50d3589_arm64-1.deb
Reading package lists... Done
E: Unsupported file ./scsi2pi_.2_devel_50d3589_arm64-1.deb given on commandline
```
Any idea why it works for you but not for me?

uweseimet, 2024-01-21:
@fdanapfel Forget my comment, I only now noticed my typo ;-). It seems to work now.

uweseimet, 2024-01-21:
@fdanapfel From your perspective, is their anything missing for closing this ticket?

I have not succeeded with setting uid root for the binaries. Even when I do that (chmod a+s) I cannot launch them as a regular user. getuid() still returns my actual user ID instead of 0. Any idea what I am missing?

fdanapfel, 2024-01-21:
@uweseimet Since installing the package via apt with all the dependecies now works I'm fine with closing the ticket.

Sorry, no idea about the setuid()/getuid() stuff. Have to do some research to see if there is a way to run binaries that access GPIO on the Pi as a normal user without using sudo.

uweseimet, 2024-01-21:
@fdanapfel Closing thist ticket, thank you for your help.

closed 2024-01-21 by uweseimet

uweseimet, 2024-02-23:
@fdanapfel Just for the sake of completeness: With the latest SCSI2Pi 3.0 binary builds sudo is not required anymore for launching s2p or the other tools, provided that a binary build was installed or "sudo make install" was used to install the self-compiled binaries.