uweseimet/scsi2pi #36: s2pctl crashes with segmentation fault in interactive mode when "enter" key is pressed without any other input

fdanapfel, 2024-01-22:
# When creating a new ticket please provide information on your environment.

- The Pi type:
Zero 2 W

- The Pi OS version (output of 'lsb_release -a'):
Distributor ID: Debian
Description: Debian GNU/Linux 12 (bookworm)
Release: 12
Codename: bookworm

- The SCSI2Pi release or git revision:
scsi2pi_1.2_devel_50d3589_arm64-1.deb

- The computer/sampler connected to the RaSCSI/PiSCSI board:
N/A

# Describe the issue
When running s2pctl in interactive mode and just pressing enter without entering anything else it crashes with a "Segmentation fault":
```
$ /opt/scsi2pi/bin/s2pctl 
SCSI Target Emulator and SCSI Tools SCSI2Pi (Server Controller Tool)
Version 1.2-devel
Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2023 Contributors to the PiSCSI project
Copyright (C) 2021-2024 Uwe Seimet
Entering interactive mode, Ctrl-D or "exit" to quit
s2pctl>
Segmentation fault
```

When entering commands it works without issues:
```
$ /opt/scsi2pi/bin/s2pctl 
SCSI Target Emulator and SCSI Tools SCSI2Pi (Server Controller Tool)
Version 1.2-devel
Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2023 Contributors to the PiSCSI project
Copyright (C) 2021-2024 Uwe Seimet
Entering interactive mode, Ctrl-D or "exit" to quit
s2pctl>server-version
Device server version: SCSI2Pi 1.2-devel
s2pctl>
```

uweseimet, 2024-01-22:
@fdanapfel Indeed ;-). For the sake of simplicity I will fix that while working on https://github.com/uweseimet/scsi2pi/issues/37.

uweseimet, 2024-01-22:
@fdanapfel The new issue_37 banch should fix this issue. Can you please test?

fdanapfel, 2024-01-22:
@uweseimet I prefer to use the binary packages, and since this is not a major issue I'm happy to wait until the fix is included in a binary package to test if the issue is fixed.

uweseimet, 2024-01-23:
@fdanapfel Development builds with this issue fixed are now available for download.