uweseimet/scsi2pi #38: Systemd service to start s2p automatically during boot

fdanapfel, 2024-01-22:
# When creating a new ticket please provide information on your environment.

- The Pi type:
Zero 2 W

- The Pi OS version (output of 'lsb_release -a'):
Distributor ID: Debian
Description: Debian GNU/Linux 12 (bookworm)
Release: 12
Codename: bookworm

- The SCSI2Pi release or git revision:
scsi2pi_1.2_devel_50d3589_arm64-1.deb

- The computer/sampler connected to the RaSCSI/PiSCSI board:
N/A

# Describe the issue
Here is a systemd service file to enable the automatic start of s2p during boot (adapted from the piscsi.service file):
```
[Unit]
Description=scsi2pi s2p service
After=network.target

[Service]
Type=simple
Restart=always
ExecStart=/opt/scsi2pi/bin/s2p -r 7
# Example 1: If you want to automatically attach a hard disk at startup, 
# say an image called harddisk.hds on SCSI ID 1, change the ExecStart line to:
#
# ExecStart=/opt/scsi2pi/bin/s2p -ID1 /home/pi/images/harddisk.hds
#
# Example 2: If you want to reserve SCSI IDs to prevent usage, add '-r' followed by
# comma-separated SCSI ID numbers; for instance IDs 0 and 7:
#
# ExecStart=/opt/scsi2pi/bin/s2p -r 0,7
#
ExecStop=/opt/scsi2pi/bin/s2pctl -X
SyslogIdentifier=scsi2pi

[Install]
WantedBy=multi-user.target
```

To enable it, copy the file to /etc/systemd/system/s2p.service and then run the following commands:
```
sudo chmod 644 /etc/systemd/system/s2p.service
sudo systemctl daemon-reload
sudo systemctl start s2p.service
sudo systemctl enable s2p.service
```

uweseimet, 2024-01-22:
@fdanapfel Currently I don't want to use a system service to launch s2p because this would collide with the piscsi system service. It is more likely that somebody using s2p also uses piscsi (at least for the web UI), and less likely that somebody uses SCSI2Pi without PiSCSI.

fdanapfel, 2024-01-22:
@uweseimet Yes, I'm aware of this. I created this issue more as a reference so that others who want to use SCSI2Pi without PiSCSI can see that it's possible to also start s2p automatically on boot.

uweseimet, 2024-01-22:
@fdanapfel I see, and I will probably add this piece of information to the SCSI2Pi website. In general, I intend to add information like this, which is useful for all users, to the website. Information which is mainly relevant for developers will go to the GitHub wiki.

uweseimet, 2024-01-22:
@fdanapfel In contrast to piscsi I don't think when launched as a service s2p should have any parameters. Instead, s2p should be configured with a property file, see https://www.scsi2pi.net/en/properties.html for details.
This is a more dynamic approach, and the property file can essentially be placed anywhere. There are also other advantages. If you want to configure you own mode pages, which can be required especially for exotic platforms, you would use a property file anyway.

uweseimet, 2024-01-22:
@fdanapfel And one more thing: It's probably best to let dpkg handle parts of the s2p configuration. I assume it is possible to run "dpkg --configure" or something like that in order to configure whether you want s2p to be launched as a service. That would be similar to using raspi-config in order to configure the Pi. I have not yet investigated this, though, but if possible I want to use standard configuration mechanisms.

fdanapfel, 2024-01-22:
@uweseimet The provided s2p.service file is just a modified version of the piscsi.service file, therefore I decided to keep the same parameters in it for starting s2p that are used for starting piscsi via systemd.

Having a property file to configure parameters is probably the better option, but if s2p is run via systemd that configuration file should probably be placed in /etc and not in some users homedir by default (because systemd normally runs processes as root and you don't want to have any configuration/property files in the homedir of the root user).

I'm not that familiar with Debian/Raspberry Pi OS and therefore I don't know if it is possible to use dpkg or apt  to create the systemd services file to enable the automatic start of a service on boot. I highly doubt this is possible and  I've never seen apt ask during installation of a package if a binary provided in the package should be launched as a service or not.

On other distros (like Fedora or RHEL) the way this normally works is that the systmd service file and a file containing the default configuration are included in the package providing the binaries, and then during installation of the package the files are placed in the right location. And the package installation tool then also takes care of setting up everything to enable the automatic start of the service on boot if the packager has defined that.
If the packager has defined that the service should not be automatically enabled on boot after installation then the admin could later use tools like systemctl to start the service and also enable the automatic start on boot.

I'm guessing this is also what raspi-config does when one of the options to enable a service is chosen, it just uses tools like systemctl to enable and start it. But the .service file is most likely already installed as part of the installation of the package. Because a tool like dpkg or apt or raspi-config would not know anything abouth the specifics of how exactly a service needst to be started and stopped and what its dependencies are.

uweseimet, 2024-01-23:
@fdanapfel Creating the service during the installation is not that straightforward. I will try to address this in release 1.3, but not in 1.2 anymore. For 1.2 I want to concentrate on testing the new features.
But https://www.scsi2pi.net/en/target.html now explains how to manually set up s2p as a system service, based on SCSI2Pi 1.2.

uweseimet, 2024-01-24:
@fdanapfel Regarding the path on my Pi I actually use /etc/systemd/. Somehow In stumbled upon /lib/systemd/, but you are right, this should be changed. And yes, I missed the DaynaPort ;-). Thank you for pointing this out.

Regarding the usage of a service in general I plan to change the rules for resolving the property files. Currently the default is ~/.config/s2p.properties, but this results in an inconsistency with a service that's looking into /etc/s2p.properties. The same user will get different setups, depending on whether the service is used or s2p is launched manually.

I plan to make /etc/s2p.properties the default, which will be parsed first. ~/.config/s2p.properties will be parsed after /etc/s2p.properties. In order to be able to disable devices defined in /etc/s2p.properties I will add an additional boolean property "device.x.active", which is by default true. By setting this property to false a local property file can override anything in the global default configuration, if required.
This appraoch is also consistent  with other software that supports local defaults and global defaults in /etc.

uweseimet, 2024-01-27:
@fdanapfel I hope it's OK for you that I close this ticket. The information about the service is on the website now, and there is already https://github.com/uweseimet/scsi2pi/issues/39 as a follow-up ticket.
Note that recently I renamed the default propery file from s2p.properties to s2p.conf.

closed 2024-01-27 by uweseimet