uweseimet/scsi2pi #39: Try to make s2p installable as a system service configurable with dpkg

uweseimet, 2024-01-22:
# When creating a new ticket please provide information on your environment.

- The Pi type:

- The Pi OS version (output of 'lsb_release -a'):

- The SCSI2Pi release or git revision:

- The computer/sampler connected to the RaSCSI/PiSCSI board:

# Describe the issue

uweseimet, 2024-02-03:
@fdanapfel The latest development builds contain this change. I would appreciate your feedback.

fdanapfel, 2024-02-06:
@uweseimet Thanks for the notification. 

I've now been able to test this with scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb and as far as I can see the s2p.service file got installed as expected.

I first disabled and removed the manually created s2p.service:
```
$ sudo systemctl disable s2p.service 
Removed "/etc/systemd/system/multi-user.target.wants/s2p.service".
$ sudo systemctl stop s2p.service 
$ sudo systemctl status s2p.service 
○ s2p.service - scsi2pi s2p service
     Loaded: loaded (/etc/systemd/system/s2p.service; disabled; preset: enabled)
     Active: inactive (dead)
$ sudo rm /etc/systemd/system/s2p*
$ sudo rm /usr/lib/systemd/system/s2p*
rm: cannot remove '/usr/lib/systemd/system/s2p*': No such file or 
$ sudo systemctl daemon-reload 
$ sudo systemctl status s2p.service 
Unit s2p.service could not be found.
```
Then removed the existing scsi2pi package and installed scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb:
```
$ sudo apt remove scsi2pi
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  scsi2pi
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
(Reading database ... 81249 files and directories currently installed.)
Removing scsi2pi (2.0) ...
$ sudo apt install ./scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'scsi2pi' instead of './scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb'
The following NEW packages will be installed:
  scsi2pi
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/731 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 /home/frank/scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb scsi2pi arm64 2.1-devel [731 kB]
Selecting previously unselected package scsi2pi.
(Reading database ... 81236 files and directories currently installed.)
Preparing to unpack .../scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb ...
Unpacking scsi2pi (2.1-devel) ...
Setting up scsi2pi (2.1-devel) ...
Consider adding SCSI2Pi to your local environment with:
  export PATH=$PATH:/opt/scsi2pi/bin
  export MANPATH=$MANPATH:/opt/scsi2pi/man
N: Download is performed unsandboxed as root as file '/home/frank/scsi2pi_2.1_devel_0b10191_bookworm_arm64-1.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
```
Checked that s2p.service is available but not started and disabled:
```
$ sudo systemctl status s2p.service 
○ s2p.service - SCSI2Pi s2p service
     Loaded: loaded (/etc/systemd/system/s2p.service; disabled; preset: enabled)
     Active: inactive (dead)
```
Starting and enabling s2p.service worked as expected:
```
$ sudo systemctl start s2p.service 
$ sudo systemctl status s2p.service 
● s2p.service - SCSI2Pi s2p service
     Loaded: loaded (/etc/systemd/system/s2p.service; disabled; preset: enabled)
     Active: active (running) since Mon 2024-02-05 22:11:23 CET; 6s ago
   Main PID: 949 (s2p)
      Tasks: 2 (limit: 176)
        CPU: 30ms
     CGroup: /system.slice/s2p.service
             └─949 /opt/scsi2pi/bin/s2p

Feb 05 22:11:23 bookworm s2p[949]: SCSI Target Emulator and SCSI Tools SCSI2Pi (Device Emulation)
Feb 05 22:11:23 bookworm s2p[949]: Version 2.1-devel
Feb 05 22:11:23 bookworm s2p[949]: Copyright (C) 2016-2020 GIMONS
Feb 05 22:11:23 bookworm s2p[949]: Copyright (C) 2020-2023 Contributors to the PiSCSI project
Feb 05 22:11:23 bookworm s2p[949]: Copyright (C) 2021-2024 Uwe Seimet
Feb 05 22:11:23 bookworm s2p[949]: [2024-02-05 22:11:23.840] [info] Set log level to 'info'
Feb 05 22:11:23 bookworm s2p[949]: [2024-02-05 22:11:23.841] [info] Default image folder set to '/home/frank/images'
Feb 05 22:11:23 bookworm s2p[949]: [2024-02-05 22:11:23.844] [info] Reserved ID(s) set to 7
Feb 05 22:11:23 bookworm s2p[949]: [2024-02-05 22:11:23.844] [info] No devices currently attached.
Feb 05 22:11:23 bookworm s2p[949]: No devices currently attached.
$ sudo systemctl enable s2p.service 
Created symlink /etc/systemd/system/multi-user.target.wants/s2p.service → /etc/systemd/system/s2p.service.
```

uweseimet, 2024-02-06:
@fdanapfel Thank you for testing!

closed 2024-02-06 by uweseimet