uweseimet/scsi2pi #68: s2p: Evaluate/Implement LRU cache with configurable size

uweseimet, 2024-02-18:
It shall be evaluated whether an LRU cache results in a performance gain. If it does, it shall be implemented with a configurable size. In case it turns out that such a cache outperforms the PiSCSI legacy cache, the latter shall be removed.

bog-dan-ro, 2024-08-18:
Isn't better/easier to use mmap and let the kernel worry about caching which is quite good at?

uweseimet, 2024-08-18:
If you want to let the kernel do the work just use one of the non-legacy caching modes, which are filesystem based.

bog-dan-ro, 2024-08-18:
Is not the same thing, each read/write operations requires context switches which are quite expensive...
I forgot to mention that `mmap` will NOT work for 32bit systems as it won't be able to map files over 2GB, so it's probably a bad idea anyway ...