uweseimet/scsi2pi #72: Add SCSI2Pi version to the INQUIRY data

uweseimet, 2024-02-23:
As long as a user has not modified the INQUIRY product data one can easily see the current SCSI2Pi version in the revision part of the emulated device. This is not the case for non-default product data, because a user may have configured custom revision data, or with the DaynaPort emulation.
In order for the version information to always be available it shall be added to the additional response data of the INQUIRY command. This change is SCSI-compliant. Plextor drives, for instance, also do that:
```
s2pexec>-i 5:0 -c 12:00:00:00:40:00
00000000  05:80:02:02:33:00:00:18:50:4c:45:58:54:4f:52:20  '....3...PLEXTOR '
00000010  43:44:2d:52:4f:4d:20:50:58:2d:34:30:54:53:20:20  'CD-ROM PX-40TS  '
00000020  31:2e:31:31:30:32:2f:32:31:2f:30:30:20:30:31:3a  '1.1102/21/00 01:'
00000030  31:31:20:20:20:20:20:20                          '11      '
```

uweseimet, 2024-02-24:
```
00000000  03:00:05:02:3f:00:00:00:53:43:53:49:32:50:69:20  '....?...SCSI2Pi '
00000010  48:6f:73:74:20:53:65:72:76:69:63:65:73:20:20:20  'Host Services   '
00000020  30:33:31:30:53:43:53:49:32:50:69:20:33:2e:31:2d  '0310SCSI2Pi 3.1-'
00000030  64:65:76:65:6c:20:28:43:29:20:55:77:65:20:53:65  'devel (C) Uwe Se'
00000040  69:6d:65:74                                      'imet'
```

uweseimet, 2024-02-24:
@fdanapfel It would be great if you could help with testing this change on a Mac. The DaynaPort driver for the Mac has non-standard expectations about how certain SCSI commands work. This driver might have an issue with this change, but I cannot really test this. In case you have time for it, can you please check whether with the Mac the DaynaPort emulation still works? The latest develop (but not the 3.0-rc binaries) binaries already contain this change, e.g. https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_dd91adf_bookworm_arm64-1.deb.

fdanapfel, 2024-02-25:
@uweseimet I've now tried [scsi2pi_3.1_devel_dd91adf_bookworm_arm64-1.deb](https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_dd91adf_bookworm_arm64-1.deb), but with this version my Mac SE/30 no longer "sees" the DaynaPort device.
After switching back to an older develop version of scsi2pi (scsi2pi_2.1_devel_20d989b_bookworm_arm64-1.deb) the DaynaPort is visible again and the Mac is able to access the network.

uweseimet, 2024-02-25:
@fdanapfel Thank you for testing. Can you please also test with the SCSI2Pi 3.0 rc6 release candidate? This one does not contain the respective change, i.e. it should work.

fdanapfel, 2024-02-25:
@uweseimet With scsi2pi_3.0_rc6_67f83ef_bookworm_arm64-1.deb the Mac is still able to "see" the Daynaport.

uweseimet, 2024-02-25:
@fdanapfel The DaynaPort driver for the Mac is really a bad piece of software :-(. Fortunately the feature this ticket is dealing with is not important. I will discard it. The next development build (available in about an hour) will not contain it anymore.
Thank you for testing.

closed 2024-02-25 by uweseimet

fdanapfel, 2024-02-25:
@uweseimet You're welcome. Glad I could help (even though the result wasn't what you expected).

uweseimet, 2024-02-25:
@fdanapfel Well, let's say I was not that surprised that this driver would cause an issue. In case you have the time, please double-check whether the next development build (it will be https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_c4bc4ca_bookworm_arm64-1.deb in about an hour) works again.

fdanapfel, 2024-02-25:
@uweseimet Yepp, https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_c4bc4ca_bookworm_arm64-1.deb works again.