uweseimet/scsi2pi #75: Can't set IP address when attaching DaynaPort device

fdanapfel, 2024-02-25:
# When creating a new ticket please provide information on your environment.

- The Pi type:
Zero 2 W

- The Pi OS version (output of 'lsb_release -a' and 'uname -a'):
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
Linux bookworm 6.1.0-rpi8-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.73-1+rpt1 (2024-01-25) aarch64 GNU/Linux

- The SCSI2Pi release or git revision:
scsi2pi_3.1_devel_c4bc4ca_bookworm_arm64-1.deb

- The computer/sampler connected to the RaSCSI/PiSCSI board:
N/A

# Describe the issue

When attaching the DaynaPort and trying to set the IP address the IP address is ignored and 10.10.20.1/24 is always used instead:
```
$ s2pctl -l
No devices currently attached.
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:78:10:a2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.23/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 846068sec preferred_lft 846068sec
    inet6 2003:c9:d701:1200:470a:d09a:e629:147e/64 scope global dynamic noprefixroute 
       valid_lft 6991sec preferred_lft 3391sec
    inet6 fe80::8d9b:2326:42e9:e441/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
$ s2pctl -i 6 -c a -t scdp inet=192.168.178.23:interface=wlan0
$ s2pctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
frank@bookworm:~ $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:78:10:a2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.23/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 846040sec preferred_lft 846040sec
    inet6 2003:c9:d701:1200:470a:d09a:e629:147e/64 scope global dynamic noprefixroute 
       valid_lft 6963sec preferred_lft 3363sec
    inet6 fe80::8d9b:2326:42e9:e441/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
15: piscsi0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 0e:83:50:de:cf:d4 brd ff:ff:ff:ff:ff:ff
    inet 10.10.20.1/24 brd 10.10.20.255 scope global piscsi0
       valid_lft forever preferred_lft forever
    inet6 fe80::c83:50ff:fede:cfd4/64 scope link 
       valid_lft forever preferred_lft forever
$ 
```

uweseimet, 2024-02-25:
@fdanapfel This only happens with you attach the device with s2pctl, but not when launching s2p with the respective parameters?

uweseimet, 2024-02-25:
@fdanapfel I see what's wrong. s2pctl works similar to scsictl (piscsi): Where in s2p/piscsi you pass a filename, in s2pctl you have to use the "-f" option. This means in your case the correct command is:
```
>s2pctl -i 6 -c a -t scdp -f inet=192.168.178.23:interface=wlan0
```

fdanapfel, 2024-02-25:
@uweseimet Ah, OK. With `-f` it works:
```
$ s2pctl -i 6 -c a -t scdp -f inet=192.168.178.23:interface=wlan0
$ s2pctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  0 |   0 | SCHD | /home/frank/images/HD00_Atari_EasyMiNT.hda
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether e4:5f:01:78:10:a2 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.23/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 860856sec preferred_lft 860856sec
    inet6 2003:c9:d701:1200:470a:d09a:e629:147e/64 scope global dynamic noprefixroute 
       valid_lft 6773sec preferred_lft 3173sec
    inet6 fe80::8d9b:2326:42e9:e441/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
18: piscsi0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether de:4b:f2:db:17:27 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.23/24 brd 192.168.178.255 scope global piscsi0
       valid_lft forever preferred_lft forever
    inet6 fe80::dc4b:f2ff:fedb:1727/64 scope link 
       valid_lft forever preferred_lft forever
```

I tried to use your example from https://github.com/uweseimet/scsi2pi/issues/64#issuecomment-1934738178 where the `-f` was missing (because you used s2p there) as well and therefore assumed that it wasn't needed for s2pctl either.

Maybe you could add a message that notifies users in case they try to give extra parameters without putting `-f` in front.

uweseimet, 2024-02-25:
@fdanapfel I think it's better that if there is a free parameter but no -f option to use the free parameter as if it had been specified with -f. I will apply this change for the next builds.
Note, though, that using s2pctl this way is incompatible with PiSCSI's scsictl. Regarding a message for the user, you should request this change for PiSCSI, because if scsictl stays like it is it will have exactly the behavior you want to be changed.

uweseimet, 2024-02-25:
@fdanapfel Development builds with the change mentioned above are now available. It's your turn to test whether it works and does not have any unwanted side-effects ;-).
Like with s2p you can now also use shortcuts for the device type (as long as there are no other parameters) instead of specifying the device type with -t, e.g.
```
>s2pctl -i 6 -c a daynaport
```
or the filename for devices with image file, e.g.
```
>s2pctl -i 0 -c a image.hds
```

closed 2024-02-26 by uweseimet