uweseimet/scsi2pi #84: DaynaPort emulation: Improve support for parprouted

uweseimet, 2024-03-13:
s2p shall be extended to better support parprouted. (Also see https://github.com/uweseimet/scsi2pi/issues/80 and https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link#wireless-raspberry-pi-setup).

uweseimet, 2024-03-13:
@fdanapfel This is a bit tricky, because if https://stackoverflow.com/questions/41678219/how-to-properly-put-network-interface-into-promiscuous-mode-on-linux is correct you cannot see whether promiscuous mode was enabled programmatically with 'ifconfig' or 'ip'.
Because you know this topic well and have a working setup, can you please test whether the development build https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_e99c909_bookworm_arm64-1.deb successfully enables it? For this build the bridge is off by default, i.e. the log on trace level should show "Enabling promiscuous mode for eth0". No error should be reported. You should just have to manually enable IPv4 forwarding.

Regarding IP forwarding I don't think that this should be dealt with by s2p because it affects the Pi globally. s2p should not become intrusive, in particular because it may not be the only application running on the Pi. Imagine any application that needs IP forwarding enables/disables it individually. The system would become unusable or at least you would not know anymore what the actual network settings are. They might change any time and the setting in /etc/sysctl.conf loses its meaning.
s2p should just deal with settings that only affect s2p and do not , e.g. as a side-effect, also affect other applications. Even enabling promiscuous mode is already quite intrusive.

uweseimet, 2024-03-17:
@fdanapfel I would appreciate your feedback.

fdanapfel, 2024-03-18:
@uweseimet Sorry, was busy with other stuff last week.

But I now managed to test https://www.scsi2pi.net/packages/develop/scsi2pi_3.1_devel_f9af4b7_bookworm_arm64-1.deb and here is the result:
```
$ sudo apt install ./scsi2pi_3.1_devel_e99c909_bookworm_arm64-1.deb
...
Preparing to unpack .../scsi2pi_3.1_devel_e99c909_bookworm_arm64-1.deb ...
Unpacking scsi2pi (3.1-devel) ...
Setting up scsi2pi (3.1-devel) ...
Add the SCSI2Pi tools and manpages to your environment with:
  export PATH=$PATH:/opt/scsi2pi/bin
  export MANPATH=$MANPATH:/opt/scsi2pi/man
e.g. in your .bash_profile
$ sudo systemctl start s2p.service 
$ sudo systemctl status s2p.service 
● s2p.service - SCSI2Pi s2p service
     Loaded: loaded (/etc/systemd/system/s2p.service; disabled; preset: enabled)
     Active: active (running) since Mon 2024-03-18 11:45:01 CET; 7s ago
   Main PID: 954 (s2p)
      Tasks: 2 (limit: 174)
        CPU: 36ms
     CGroup: /system.slice/s2p.service
             └─954 /opt/scsi2pi/bin/s2p

Mar 18 11:45:01 bookworm s2p[954]: SCSI Target Emulator and SCSI Tools SCSI2Pi (Device Emulation)
Mar 18 11:45:01 bookworm s2p[954]: Version 3.1-devel
Mar 18 11:45:01 bookworm s2p[954]: Copyright (C) 2016-2020 GIMONS
Mar 18 11:45:01 bookworm s2p[954]: Copyright (C) 2020-2023 Contributors to the PiSCSI project
Mar 18 11:45:01 bookworm s2p[954]: Copyright (C) 2021-2024 Uwe Seimet
Mar 18 11:45:01 bookworm s2p[954]: [2024-03-18 11:45:01.516] [info] Set log level to 'info'
Mar 18 11:45:01 bookworm s2p[954]: [2024-03-18 11:45:01.521] [info] Default image folder set to '/home/frank/images'
Mar 18 11:45:01 bookworm s2p[954]: [2024-03-18 11:45:01.533] [info] Reserved ID(s) set to 7
Mar 18 11:45:01 bookworm s2p[954]: [2024-03-18 11:45:01.534] [info] No devices currently attached.
Mar 18 11:45:01 bookworm s2p[954]: No devices currently attached.
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:d8:1b:a4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.43/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863583sec preferred_lft 863583sec
    inet6 2003:c9:d701:1200:470a:d09a:e629:147e/64 scope global dynamic noprefixroute 
       valid_lft 6786sec preferred_lft 3186sec
    inet6 fe80::8d9b:2326:42e9:e441/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
$ s2pctl -l
No devices currently attached.
$ s2pctl -i 6 -c a daynaport
$ s2pctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
$ ip a|grep PROMISC
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:d8:1b:a4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.43/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863527sec preferred_lft 863527sec
    inet6 2003:c9:d701:1200:470a:d09a:e629:147e/64 scope global dynamic noprefixroute 
       valid_lft 6730sec preferred_lft 3130sec
    inet6 fe80::8d9b:2326:42e9:e441/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: piscsi0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether f6:84:c3:0a:c4:5d brd ff:ff:ff:ff:ff:ff
    inet 10.10.20.1/24 brd 10.10.20.255 scope global piscsi0
       valid_lft forever preferred_lft forever
    inet6 fe80::f484:c3ff:fe0a:c45d/64 scope link 
       valid_lft forever preferred_lft forever
```
As far as I can see promiscuous mode did not get enabled for the wlan0 interface.

If I manually enable promiscuous mode for the wlan0 interface it shows up in both the output of the `ip a` command and also in dmesg:
```
$ sudo /sbin/ip link set wlan0 promisc on
$ dmesg|grep -i promisc
[  852.008786] brcmfmac mmc1:0001:1 wlan0: entered promiscuous mode
$ ip a|grep PROMISC
2: wlan0: <BROADCAST,MULTICAST,PROMISC,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
$ 
```
Maybe you can check with strace what the ip command is doing to enable promiscous mode for an interface and then use the same calls in s2p.

Regarding your concerns about having s2p deal with things like enabling IP forwarding or setting promiscous mode on an interface, while I agree with this to some degree I don't think it should be an issue for this kind of setup. The reason being that I don't think there will ever be any other application running on this kind of setup that also wants to make changes to these settings, and as already mentioned at least setting IP forwarding is already required when using the current NAT setup used by PiSCSI for the WLAN setup.
Also when setting up a bridge, the interface that is used for one end of the bridge is also put in promiscuous mode, so for a setup where the piscsi_bridge is used this is already happening when the bridge is created.

IMHO it is better to only have IP. forwarding and promiscous mode enabled while the daynaport adapter is enabled, instead of having them activated permanently. Having users deal with those setups outside of s2p in my opinion makes the whole setup more error prone.

uweseimet, 2024-03-18:
@fdanapfel Thank you for testing.
Regarding changing IP forwarding I have decided not to touch it for the reasons I have already mentioned. In addition, there is no C/C++ API for dealing with IP forwarding (something ioctl based) as far as I can tell. A missing API is also a strong indication that these settings are not meant to be changed programmatically.
I will try to check programmatically whether ioctl says that promiscuous mode is enabled. It might turn out that enabling it programmatically is not the same as doing this in the command line. If this is the case I am afraid I most likely will have to close this ticket unresolved.
There is only scarce information on all this, which shows that hardly anybody is changing anything programmatically, probably for good reasons.

fdanapfel, 2024-03-18:
@uweseimet I did a quick google search and found the following which might help on how to enable promiscuous mode in C:
https://www.cs.usfca.edu/~cruse/cs686s08/netsniff.cpp
https://github.com/bsdb0y/Raw-Packet-Sniffer-in-C/blob/master/raw_sock_in.c
Maybe this helps. And lastly you could also try to take a look at the code of the `ip` command, because after all "doing this in the command line" is just calling a binary, which I'm guessing has also been written in C/C++

uweseimet, 2024-03-19:
After reading more about promiscuous mode I decided not to fiddle with it in s2p. s2p is not a tool like wireshark or the pcap library, which have a special interest in this mode. Manipulating it in a more or less normal application may have unwanted side effects.

closed 2024-03-19 by uweseimet