uweseimet/scsi2pi #92: Amiga - CD Rom Devices do not work

SavourySnaX, 2024-07-22:
# When creating a new ticket please provide information on your environment.

- The Pi type:
Pi Zero

- The Pi OS version (output of 'lsb_release -a' and 'uname -a'):
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

- The SCSI2Pi release or git revision:
3.4 Release

- The computer/sampler connected to the RaSCSI/PiSCSI board:
Amiga 1200 - Via Blizzard SCSI Kit IV with 8.5 Rom

# Describe the issue

Tried installing the 3.4 release to see if it was a suitable replacement for the piscsi v24.04.01 release, I have a lot of devices in the config for the device. With 2 cd roms (Units 6 0 and 6 1), on the piscsi drivers the 2 cdroms are mounted and behave correctly, on the scsi2pi 3.4 release, the cd roms mount incorrectly, 1 detects as audio cd (its purely data), the other behaves even more strangely (and causes standard amiga assigns to be lost). reverting back to piscsi and the devices return to normal behaviour. 

I can do further testing, if there is something specific i can enable to trace the issues i may be willing. I've only tried 3.4, I could bisect the releases if it would help. My time is limited however, so it might be a few weekends before I can do that.

uweseimet, 2024-07-22:
Bisecting would not really help because the SCSI2Pi repository does not contain the legacy PiSCSI codebase, and due to numerous new features and cleanups it differs quite a lot from PiSCSI anyway.
I don't have an Amiga, but I might be able to figure out what's wrong if you can provide the images of your CD-ROMs. One image is enough for a start, because a step-by-step approach is best. I would also be interested in logs showing what happens when you mount/access this image with PiSCSI and with SCSI2Pi.
When testing, please only attach one image, not both. When the issue is resolved for one image, the other can be addressed. Otherwise the test setup and the logs become too complicated.

uweseimet, 2024-07-22:
Instead of providing an image, running this command on your image file and attaching the output may already be sufficient:
```
od --endian=big -v -Ax -x -N 4096 NAME_OF_YOUR_IMAGE_FILE
```

SavourySnaX, 2024-07-22:
The output from the above command is a file with all 00's in it, these are straightforward .iso images, the first 32k in both cases are 0's (ISO_9660), so I dumped the first 64k from one, in case that is more useful.
[test.zip](https://github.com/user-attachments/files/16335802/test.zip)

I`ll try and get the logs shortly.

SavourySnaX, 2024-07-22:
Trace logs for scsi2pi : 

```
[2024-07-22 16:21:00.440] [debug] (ID:LUN 6:0) - Device is executing TEST UNIT READY ($00)
[2024-07-22 16:21:00.442] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:00.442] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.442] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.443] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:00.443] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.444] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.447] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:21:00.448] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:21:00.449] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:21:00.449] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:21:00.450] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:21:00.450] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:21:00.451] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:21:00.451] [debug] (ID 6) - Controller is executing READ TOC, CDB $43:02:02:00:00:00:00:04:50:00
[2024-07-22 16:21:00.451] [debug] (ID:LUN 6:0) - Device is executing READ TOC ($43)
[2024-07-22 16:21:00.453] [trace] (ID 6) - DATA IN phase
[2024-07-22 16:21:00.454] [trace] (ID 6) - Sending 12 byte(s)
[2024-07-22 16:21:00.454] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.455] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:00.456] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.457] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.457] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:00.459] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.460] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.461] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:21:00.463] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:21:00.463] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:21:00.464] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:21:00.464] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:21:00.464] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:21:00.465] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:21:00.465] [debug] (ID 6) - Controller is executing READ TOC, CDB $43:00:00:00:00:00:00:03:24:00
[2024-07-22 16:21:00.466] [debug] (ID:LUN 6:0) - Device is executing READ TOC ($43)
[2024-07-22 16:21:00.467] [trace] (ID 6) - DATA IN phase
[2024-07-22 16:21:00.468] [trace] (ID 6) - Sending 12 byte(s)
[2024-07-22 16:21:00.470] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.470] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:00.471] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.473] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.473] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:00.475] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.475] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.475] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:21:00.480] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:21:00.480] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:21:00.480] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:21:00.480] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:21:00.480] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:21:00.481] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:21:00.481] [debug] (ID 6) - Controller is executing READ TOC, CDB $43:00:05:00:00:00:00:07:1e:00
[2024-07-22 16:21:00.482] [debug] (ID:LUN 6:0) - Device is executing READ TOC ($43)
[2024-07-22 16:21:00.482] [trace] (ID 6) - DATA IN phase
[2024-07-22 16:21:00.483] [trace] (ID 6) - Sending 12 byte(s)
[2024-07-22 16:21:00.484] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.485] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:00.486] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.487] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.488] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:00.489] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:00.489] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:00.490] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:21:02.164] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:21:02.164] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:21:02.165] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:21:02.165] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:21:02.165] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[... 16 lines cut ...]
[2024-07-22 16:21:05.187] [debug] (ID 6) - Controller is executing TEST UNIT READY, CDB $00:00:00:00:00:00
[2024-07-22 16:21:05.187] [debug] (ID:LUN 6:0) - Device is executing TEST UNIT READY ($00)
[2024-07-22 16:21:05.187] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:05.188] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:05.189] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:05.191] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:05.192] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:05.193] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:05.194] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:21:08.204] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:21:08.204] [trace] (ID 6) - SELECTION phase
[... cut ...]
[2024-07-22 16:21:08.205] [debug] (ID:LUN 6:0) - Device is executing TEST UNIT READY ($00)
[2024-07-22 16:21:08.205] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:21:08.205] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:08.205] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:08.205] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:21:08.205] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:21:08.205] [trace] (ID 6) - All data transferred
[2024-07-22 16:21:08.205] [trace] (ID 6) - BUS FREE phase
```

uweseimet, 2024-07-22:
Thank you. I would also like to request the piscsi trace logs for the same operation.

SavourySnaX, 2024-07-22:
piscsi version :

```
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.951] [trace] (ID:LUN 6:0) - Device is ready
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - Message In phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.952] [trace] (ID 6) - Bus Free phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Selection phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Selection completed
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Message Out phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Receiving data, transfer length: 1 byte(s)
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Phase: msgout
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Received IDENTIFY message for LUN 0
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.953] [trace] (ID 6) - Command phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.954] [trace] (ID 6) - Controller is executing ReadToc, CDB $43020200000000045000
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.954] [debug] (ID:LUN 6:0) - Device is executing ReadToc ($43)
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.954] [trace] (ID:LUN 6:0) - Device is ready
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.954] [trace] (ID 6) - Data In phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.954] [trace] (ID 6) - Sending data, offset: 0, length: 1104
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - All data transferred, moving to next phase: datain
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - Message In phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.959] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.960] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.960] [trace] (ID 6) - Bus Free phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.960] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.960] [trace] (ID 6) - Selection phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.960] [trace] (ID 6) - Selection completed
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Message Out phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Receiving data, transfer length: 1 byte(s)
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Phase: msgout
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Received IDENTIFY message for LUN 0
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Command phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Controller is executing ReadToc, CDB $43000000000000032400
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [debug] (ID:LUN 6:0) - Device is executing ReadToc ($43)
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID:LUN 6:0) - Device is ready
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Data In phase
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.961] [trace] (ID 6) - Sending data, offset: 0, length: 804
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.965] [trace] (ID 6) - All data transferred, moving to next phase: datain
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.965] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.965] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.966] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:25:33 raspberrypi PISCSI[1008]: [2024-07-22 16:25:32.966] [trace] (ID 6) - Message In phase
[... cut ...]
Jul 22 16:25:38 raspberrypi PISCSI[1008]: [2024-07-22 16:25:38.319] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:25:38 raspberrypi PISCSI[1008]: [2024-07-22 16:25:38.319] [trace] (ID 6) - Message In phase
Jul 22 16:25:38 raspberrypi PISCSI[1008]: [2024-07-22 16:25:38.319] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:25:38 raspberrypi PISCSI[1008]: [2024-07-22 16:25:38.320] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:25:38 raspberrypi PISCSI[1008]: [2024-07-22 16:25:38.320] [trace] (ID 6) - Bus Free phase
^C
```

uweseimet, 2024-07-22:
Thank you. Looks as if SCSI2Pi stops early when reading the TOC.

Any change that you upload the full ISO image somewhere? If needed can probably provide a temporary sftp location for this purpose.

SavourySnaX, 2024-07-22:
Let me see if i can find a cd that I can share - (amigaos3.2 is paid and non distributable)

[EasyNet (2010-06-10)(AmigaKit).zip](https://github.com/user-attachments/files/16336315/EasyNet.2010-06-10.AmigaKit.zip)
This image exhibits the same problem.  (perhaps its something during boot - the amiga probes the devices during early config) - I`ll attach full logs of upto mounting in a second.

SavourySnaX, 2024-07-22:
s2p

```
[2024-07-22 16:34:21.977] [info] Set log level to 'trace'
[2024-07-22 16:34:22.024] [trace] Executing CHECK_AUTHENTICATION command
[2024-07-22 16:34:22.024] [trace] Received CHECK_AUTHENTICATION command
[2024-07-22 16:34:22.036] [trace] Executing SERVER_INFO command
[2024-07-22 16:34:22.153] [trace] Executing DEVICES_INFO command
[2024-07-22 16:34:22.166] [trace] Executing DEFAULT_IMAGE_FILES_INFO command
[2024-07-22 16:34:22.189] [trace] Executing DEVICE_TYPES_INFO command
[2024-07-22 16:34:22.220] [trace] Executing DEFAULT_IMAGE_FILES_INFO command
[2024-07-22 16:34:22.242] [trace] Executing SERVER_INFO command
[2024-07-22 16:34:22.388] [trace] Executing NETWORK_INTERFACES_INFO command
[2024-07-22 16:34:22.438] [trace] Executing DEVICE_TYPES_INFO command
[2024-07-22 16:34:22.466] [trace] Executing DEVICE_TYPES_INFO command
[2024-07-22 16:34:22.492] [trace] Executing DEVICE_TYPES_INFO command
[2024-07-22 16:34:22.517] [trace] Executing DEVICE_TYPES_INFO command
[2024-07-22 16:34:22.545] [trace] Executing SERVER_INFO command
[2024-07-22 16:34:23.172] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:34:23.173] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:34:23.173] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:34:23.173] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:34:23.173] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:34:23.173] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:34:23.173] [debug] (ID 6) - Controller is executing TEST UNIT READY, CDB $00:00:00:00:00:00
[2024-07-22 16:34:23.173] [debug] (ID:LUN 6:0) - Device is executing TEST UNIT READY ($00)
[2024-07-22 16:34:23.173] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:34:23.173] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.173] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.173] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:34:23.173] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.174] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.174] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:34:23.175] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:34:23.176] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:34:23.178] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:34:23.178] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:34:23.179] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:34:23.180] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:34:23.180] [debug] (ID 6) - Controller is executing READ CAPACITY(10), CDB $25:00:00:00:00:00:00:00:00:00
[2024-07-22 16:34:23.181] [debug] (ID:LUN 6:0) - Device is executing READ CAPACITY(10) ($25)
[2024-07-22 16:34:23.181] [trace] (ID 6) - DATA IN phase
[2024-07-22 16:34:23.181] [trace] (ID 6) - Sending 8 byte(s)
[2024-07-22 16:34:23.182] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.183] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:34:23.193] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.194] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.195] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:34:23.195] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.195] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.196] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:34:23.197] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:34:23.198] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:34:23.198] [trace] (ID 6) - MESSAGE OUT phase
[2024-07-22 16:34:23.198] [trace] (ID 6) - Receiving 1 byte(s)
[2024-07-22 16:34:23.200] [trace] (ID 6) - Received IDENTIFY message for LUN 0
[2024-07-22 16:34:23.200] [trace] (ID 6) - COMMAND phase
[2024-07-22 16:34:23.200] [debug] (ID 6) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:0c:00
[2024-07-22 16:34:23.201] [debug] (ID:LUN 6:0) - Device is executing MODE SENSE(6) ($1a)
[2024-07-22 16:34:23.202] [debug] (ID 6) - ILLEGAL REQUEST (Sense Key $05), INVALID FIELD IN CDB (ASC $24), ASCQ $00
[2024-07-22 16:34:23.203] [trace] (ID 6) - Status phase, status is CHECK CONDITION (status code $02)
[2024-07-22 16:34:23.204] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.204] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.205] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:34:23.206] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:34:23.206] [trace] (ID 6) - All data transferred
[2024-07-22 16:34:23.207] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:34:23.208] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:34:23.208] [trace] (ID 6) - SELECTION phase
[2024-07-22 16:34:23.208] [trace] (ID 6) - MESSAGE OUT phase
[... 665 lines cut ...]
[2024-07-22 16:35:58.928] [debug] (ID:LUN 6:0) - Device is executing TEST UNIT READY ($00)
[2024-07-22 16:35:58.928] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:35:58.928] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:35:58.929] [trace] (ID 6) - All data transferred
[2024-07-22 16:35:58.930] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:35:58.930] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:35:58.933] [trace] (ID 6) - All data transferred
[2024-07-22 16:35:58.933] [trace] (ID 6) - BUS FREE phase
[2024-07-22 16:36:01.945] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2024-07-22 16:36:01.945] [trace] (ID 6) - SELECTION phase
[... cut ...]
[2024-07-22 16:36:01.947] [trace] (ID 6) - Status phase, status is GOOD (status code $00)
[2024-07-22 16:36:01.947] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:36:01.948] [trace] (ID 6) - All data transferred
[2024-07-22 16:36:01.948] [trace] (ID 6) - MESSAGE IN phase
[2024-07-22 16:36:01.948] [trace] (ID 6) - Sending 1 byte(s)
[2024-07-22 16:36:01.949] [trace] (ID 6) - All data transferred
[2024-07-22 16:36:01.949] [trace] (ID 6) - BUS FREE phase
^C[2024-07-22 16:36:02.176] [info] Detached all devices
```

SavourySnaX, 2024-07-22:
piscsi
```
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [trace] (ID 6) - Controller is executing TestUnitReady, CDB $000000000000
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [debug] (ID:LUN 6:0) - Device is executing TestUnitReady ($00)
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [trace] (ID:LUN 6:0) - Device is ready
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.766] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.767] [trace] (ID 6) - Message In phase
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.767] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.767] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:41:15 raspberrypi PISCSI[1245]: [2024-07-22 16:41:15.767] [trace] (ID 6) - Bus Free phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.577] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.577] [trace] (ID 6) - Selection phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.577] [trace] (ID 6) - Selection completed
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.577] [trace] (ID 6) - Command phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - Controller is executing TestUnitReady, CDB $000000000000
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [debug] (ID:LUN 6:0) - Device is executing TestUnitReady ($00)
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID:LUN 6:0) - Device is ready
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - Message In phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.578] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.579] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.579] [trace] (ID 6) - Bus Free phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.579] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Selection phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Selection completed
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Command phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Controller is executing Inquiry, CDB $12000000fe00
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [debug] (ID:LUN 6:0) - Device is executing Inquiry ($12)
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Data In phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.580] [trace] (ID 6) - Sending data, offset: 0, length: 36
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - All data transferred, moving to next phase: datain
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - Status phase, status is $00
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - Message In phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.581] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.582] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.582] [trace] (ID 6) - Bus Free phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [trace] (ID 6) - Selection phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [trace] (ID 6) - Selection completed
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [trace] (ID 6) - Command phase
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [trace] (ID 6) - Controller is executing StartStop, CDB $1b0000000100
Jul 22 16:41:29 raspberrypi PISCSI[1245]: [2024-07-22 16:41:29.583] [debug] (ID:LUN 6:0) - Device is executing StartStop ($1b)
[... cut ...]
Jul 22 16:42:10 raspberrypi PISCSI[1245]: [2024-07-22 16:42:10.869] [trace] (ID 6) - All data transferred, moving to next phase: status
Jul 22 16:42:10 raspberrypi PISCSI[1245]: [2024-07-22 16:42:10.869] [trace] (ID 6) - Message In phase
Jul 22 16:42:10 raspberrypi PISCSI[1245]: [2024-07-22 16:42:10.869] [trace] (ID 6) - Sending data, offset: 0, length: 1
Jul 22 16:42:10 raspberrypi PISCSI[1245]: [2024-07-22 16:42:10.869] [trace] (ID 6) - All data transferred, moving to next phase: msgin
Jul 22 16:42:10 raspberrypi PISCSI[1245]: [2024-07-22 16:42:10.869] [trace] (ID 6) - Bus Free phase
```

SavourySnaX, 2024-07-22:
Both of the above were using the image i attached just before them.

SavourySnaX, 2024-07-22:
the only pi i have that is not in use, is a 400 - it can probably do it, although its not currently plugged in - If you ping here when you have a bugfix i can probably try and compile it on there. i can't be sure when i will get chance though, will depend on work etc.

uweseimet, 2024-07-22:
Binaries for SCSI2Pi 3.4.1-devel are available now on https://www.scsi2pi.net/packages/develop/. Can you please test whether these resolve you problem, at least for one of your CD images?

SavourySnaX, 2024-07-22:
Confirmed, both cds are now reading fine. thx

uweseimet, 2024-07-22:
Great! Am I right that all of your issues are resolved with the fix I provided?

SavourySnaX, 2024-07-22:
At least as far as I can test.. I`ll do some proper digging at the weekend when I have time, but this issue can be closed.

uweseimet, 2024-07-22:
Sounds good, thank you for reporting and testing. I will most likely release SCSI2Pi 3.4.1 with this fix later today.

closed 2024-07-22 by uweseimet