uweseimet/scsi2pi #93: Streamer/Tape support

bog-dan-ro, 2024-08-18:
Same request as https://github.com/PiSCSI/piscsi/issues/480

I already have a non working WIP for piscsi (that was before I found scsi2pi). 
I'll be happy to port it to scsi2pi. But I'll someones help, who knows the internals of this project, to fix the remaining stuff.

uweseimet, 2024-08-18:
@bog-dan-ro Thank you for creating this ticket. Currently there are no plans to add streamer support to SCSI2Pi. In the past tapes were used for backups, because there was no better/cheaper way. Nowadays, especially with tools like SCSI2Pi, you simply use a drive image for a backup. This is faster and more flexible than a streamer emulation. Using a streamer emulation instead would be rather inconvenient.

However, as soon as you have added fully functional tape support to PiSCSI please get back to me. I wil then run tests with my Atari (I have streamer software for it and also an HP streamer), and I will then re-evaluate this ticket based on the results of my tests.

uweseimet, 2024-08-18:
@bog-dan-ro By the way, I would be interested to learn which methods from the Disk class you think should be moved into the StorageDevice class.

bog-dan-ro, 2024-08-18:
Actually they are not too many, check my last commit from https://github.com/bog-dan-ro/piscsi

bog-dan-ro, 2024-08-18:
See https://github.com/bog-dan-ro/piscsi/commit/52a50c91d62a574bd2bf1e8f695d74860209cd62#diff-ba201351e19eac2f5e902b9308a59f1e3bdc19ed324119dab011d8336c345410
and https://github.com/bog-dan-ro/piscsi/commit/52a50c91d62a574bd2bf1e8f695d74860209cd62#diff-76d682148e5aa7e6ce46bd9e40aae7de042a2a4e24242baf6f95b93612e6a16b
 
I didn't moved the caching as streamers do not need it as they are sequential so, once you read/write a block it will not seek back to read it again.

uweseimet, 2024-08-18:
I will have a look at the code you moved and will get back to you, but this may take some time.

Is the streamer code in your repository already tested and working?

bog-dan-ro, 2024-08-18:
> @bog-dan-ro Thank you for creating this ticket. Currently there are no plans to add streamer support to SCSI2Pi. In the past tapes were used for backups, because there was no better/cheaper solution. Nowadays, especially with tools like SCSI2Pi, you simply use a drive image for a backup. This is faster and more flexible than a streamer emulation. Using a streamer emulation instead would be rather inconvenient.

There are other use-casese for than tape than backup. For instance, Altos 386 Series 1000 needs it to install the the system utilities.
I also want to use it to transfer large software on that machine as splitting big tar files and transferring them from floppy is not fun :).

> I will have a look at the code you moved and will get back to you, but this may take some time.

No rush, I'll be in vacation for a couple of weeks.

> Is the streamer code in your repository already tested and working?

I tested it a bit, but sadly is not working ... The mandatory implementation is there, but I don't know how to properly debug it. Most probably I'll have to add a modern build system (cmake?) so my IDE to know how to cross compile, deploy & debug it properly.
Also, I didn't have too much time to properly understand the API nor the workflow :). I'll have to cry for help on PiSCSI maybe someone who knows the code well, will have time to help me fix it.

I chose tandber as the reference device for two reasons:
1. Is one of the most used tape drives
2. it has proper scsi [documentation](https://bitsavers.org/pdf/tandbergData/TDC_3620_3640_3660_Reference_Manual.pdf) ;-)

uweseimet, 2024-08-18:
Indeed, I missed the use case of an installation. Anyway, I will get back to you (presumably in a couple of days) regarding moving code from Disk to StorageDevice.

There is no direct relationship between cmake and cross-compling, by the way. The nightly SCSI2Pi builds and the release builds are cross-compiled, but SCSI2Pi does not use cmake. I thought about using it, but it would not be worth the effort, considering that the Makefile is portable and straightforward.

bog-dan-ro, 2024-08-18:
> There is no direct relationship between cmake and cross-compling, by the way.

Nope is not, at least not direct. Current makefile works great to cross-compile it from command line. I created a docker file to help me cross compile it, and I have no issues to cross-compile it.
But ... Some IDEs (qtcreator) need to know what files are built, where they are on the host computer and where they should be deployed on the target.  Using CMake, gives all that info to IDE.

> Indeed, I missed the use case of an installation. Anyway, I will get back to you (presumably in a couple of days) regarding moving code from Disk to StorageDevice.

No hurry at all, starting next Friday I'll be in vacation for two weeks.

uweseimet, 2024-08-20:
@bog-dan-ro There is a new branch "code_move" now, in which I moved the code related to sector/block sizes from the Disk class to the StorageDevice class.
Changes related to shift counts (and others) are not required, because this code does not exist anymore in SCSI2Pi.

bog-dan-ro, 2024-08-22:
@uweseimet Here https://github.com/PiSCSI/piscsi/pull/1477 is the PR

uweseimet, 2024-08-22:
@bog-dan-ro Can you please remove my name in the copyright notice of the streamer code?

bog-dan-ro, 2024-08-22:
> @bog-dan-ro Can you please remove my name in the copyright notice of the new streamer class code (.cpp and .h)?

Done

uweseimet, 2024-09-25:
@bog-dan-ro In the meantime I have added a branch issue_93 with initial tape drive support to SCSI2Pi. This implementation is completely new, i.e. it is not based on any existing code.

I have already successfully tested this branch with tar, both in the tar-file compatible mode (file extension .tar), and in the full-featured mode with filemarks support. When writing files to a real device instead of the filesystem tar actually makes use of filemarks.

uweseimet, 2024-10-05:
@bog-dan-ro There has not been any recent feedback, neither in this ticket nor in https://github.com/PiSCSI/piscsi/issues/480. As far as the SCSI2Pi ticket is concerned is is essentially done and tested. What about your PiSCSI ticket/PR?
I would like to ensure that both implementations are compatible from the user's perspective. For SCSI2Pi there are binaries of the development branch with all changes included, see https://www.scsi2pi.net/packages/develop/.
See the comments in the header of tape.cpp for details on the SCSI2Pi implementation.

@Pacjunk @sidick @ppuskari Quite long ago you were interested in tape support for the PiSCSI board. In case you are still interested, I would appreciate help with testing.

ppuskari, 2024-10-05:
This is wonderful news.  I am about ready to test this after building. I have been in the middle of a major job change so have been out of pocket a bit.

Looking forward to seeing if the IIgs software can properly work with the new code and recognize it as the 3m drive and 40meg block formatted and sellable capacity.   If this works I am going to learn a lot from your code and I am very appreciative!

Petar Puskarich
***@***.***

bog-dan-ro, 2024-10-06:
Hi @uweseimet,

Sorry for slow reply, I was quite busy at work.
I'll check your implementation next weekend.
If is better than mine I'll probably back port it to piscsi if you're okay with it.

BogDan.

uweseimet, 2024-10-06:
@bog-dan-ro Backporting is fine, but what about the existing PiSCSI PR?

bog-dan-ro, 2024-10-06:
@uweseimet my patch works for me and from my POV it's done. There are a few small issues to fix on the UI side which I could not fix them myself.

uweseimet, 2024-10-06:
When will these issues be fixed? What about the device type name? Please see my last comment in the PiSCSI ticket.

Pacjunk, 2024-10-07:
> @Pacjunk @sidick @ppuskari Quite long ago you were interested in tape support for the PiSCSI board. In case you are still interested, I would appreciate help with testing.

Hi Uwe - I've been out of the loop for a fair while but I am attempting to have a look at your new code. Taking me a while to get my head around doing things with a command line!

I updated my system to the latest PiSCSI (3.5 hours!) and had it working. I then installed your package. It disabled PiSCSI (expected), but did not create a system service for scsi2pi - I did this. The web interface does not seem to work now, even though some of documentation says it will.

I added a disk and a few .TAP files to the configuration. The workstation can see the vendor details for the disk and tape drives at the console prompt - so a good start. First try booting nothing worked, second try failed during boot with all devices being detached part way through the boot. Third try and I booted the VAXstation off the disk image.  When I try and mount a tape image (which I have verified as being correct under SimH) the system just says that the device is offline. So looks like a bit more debugging is required.

So my questions are:

1 - I can't find the log file anywhere. Where is it, and what is the best way to change the logging level to trace
2 - Can I use my existing vendor properties files for the drives, or do I have to manually copy the data over to the .conf file. (maybe a few commented out, fully optioned examples in the conf file would be useful).

Thanks,

uweseimet, 2024-10-07:
@Pacjunk Thank you for your feedback!

Regarding the system service, I may have missed updating the installation webpage. Initially the PiSCSI service was shut down and the SCSi2Pi service was started up instead, when running "apt install". I removed the last step, but I have to admit I don't know why anymore ;-). I will have to revisit this.
The service file "/etc/systemd/system/s2p.service" should be there, though. Can you confirm this?

Regarding your questions:

1. Logging information is displayed with "sudo systemctl status s2p". For testing it is often easier not to run the service, but to launch s2p from the command line.
2. The vendor properties file is evaluated by the PiSCSI webservice, isn't it? I will have to check whether I can reproduce a interoperability problem with the PiSCSI webservice. Something like that probably cannot be prevented sooner or later, but I would like to keep SCSI2Pi as compatible with PiSCSI as possible. The webservice may not be able to support other (new) device types, or it might have limited support for new types.

s2p is usually configured with a property file. http://www.scsi2pi/en/properties.html. The command line options complement/override property file settings. Some configuration data, especially mode pages, require a property file. Add all available options to a sample configuration file would probably make this file quite confusing. It would also mean maintenance overhead, because with each change I would not only have to update the help texts, the manpages and the website, but also this file.

As already mentioned, for testing purposes using the command line instead of the system service is usually more convenient, as you can quickly stop s2p, change the parameters, and restart.
Booting without a single initial tape drive, and then attaching a tape image may be the best test scenario, with less complexity than the drive being present when booting. This is what I would like to suggest:

1. Boot with only a hard drive attached.
2. Then attach the tape drive, and in case the system has issues with it a log on trace level should reveal details.
3. You can assign an individual log level to the tape device with s2pctl, so that you only get trace output for your tape, but not for the hard drive.

I have tested the current develop branch with tar and mtx, but when using other (or more elaborate) tape tools it may turn out that the implementation has to be updated/changed. I think that it is compliant with the standard, but some tools may need some of the optional commands not implemented. Or bugs still have to be fixed that tar does not trigger.
I planned to test everything with a tape backup tool for the Atari, but this tool neither works with SCSi2Pi nor with my HP streamer. From the s2p logs I can see that it sends write commands with a byte count of 0 all the time. I cannot tell whether this is due to a bug in s2p or in the tool, because the real streamer also does not work.

I hope this helps to get you started.

Pacjunk, 2024-10-09:
Here is a trace of my unsuccessful attempt to mount a tape on my VAXstation. Don't know how useful this is, but anyway...

```
Oct 09 15:14:53 piscsi s2p[802]: SCSI Target Emulator and SCSI Tools SCSI2Pi (Device Emulation)
Oct 09 15:14:53 piscsi s2p[802]: Version 4.0-devel
Oct 09 15:14:53 piscsi s2p[802]: Copyright (C) 2016-2020 GIMONS
Oct 09 15:14:53 piscsi s2p[802]: Copyright (C) 2020-2023 Contributors to the PiSCSI project
Oct 09 15:14:53 piscsi s2p[802]: Copyright (C) 2021-2024 Uwe Seimet
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.738] [info] Set log level to 'info'
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.754] [info] Reserved ID(s) set to 6
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.773] [info] Executing: operation=ATTACH, device=0:0, device parameters='file=boot72.hd1'
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.775] [info] Attached SCHD 0:0
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.777] [info] Executing: operation=ATTACH, device=3:0, device parameters='file=vms_v5.4_1.tap'
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.780] [info] Attached SCTP 3:0
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.782] [info] Executing: operation=ATTACH, device=4:0, device parameters='file=vms_v5.4_2.tap'
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.783] [info] Attached SCTP 4:0
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.783] [info] Executing: operation=ATTACH, device=5:0, device parameters='file=Test.tap'
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.784] [info] Attached SCTP 5:0
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] +----+-----+------+-------------------------------------
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] | ID | LUN | TYPE | IMAGE FILE
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] +----+-----+------+-------------------------------------
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] |  0 |   0 | SCHD | /home/pi/images/boot72.hd1
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] |  3 |   0 | SCTP | /home/pi/images/vms_v5.4_1.tap
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] |  4 |   0 | SCTP | /home/pi/images/vms_v5.4_2.tap
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] |  5 |   0 | SCTP | /home/pi/images/Test.tap
Oct 09 15:14:53 piscsi s2p[802]: [2024-10-09 15:14:53.785] [info] +----+-----+------+-------------------------------------
Oct 09 15:14:53 piscsi s2p[802]: +----+-----+------+-------------------------------------
Oct 09 15:14:53 piscsi s2p[802]: | ID | LUN | TYPE | IMAGE FILE
Oct 09 15:14:53 piscsi s2p[802]: +----+-----+------+-------------------------------------
Oct 09 15:14:53 piscsi s2p[802]: |  0 |   0 | SCHD | /home/pi/images/boot72.hd1
Oct 09 15:14:53 piscsi s2p[802]: |  3 |   0 | SCTP | /home/pi/images/vms_v5.4_1.tap
Oct 09 15:14:53 piscsi s2p[802]: |  4 |   0 | SCTP | /home/pi/images/vms_v5.4_2.tap
Oct 09 15:14:53 piscsi s2p[802]: |  5 |   0 | SCTP | /home/pi/images/Test.tap
Oct 09 15:14:53 piscsi s2p[802]: +----+-----+------+-------------------------------------
Oct 09 15:17:02 piscsi s2p[802]: [2024-10-09 15:17:02.554] [info] Set log level to 'trace'
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - ++++ Starting processing for initiator ID 6
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - SELECTION phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - MESSAGE OUT phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - Receiving 1 byte(s)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - Received IDENTIFY message for LUN 0
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.284] [trace] (ID 3) - COMMAND phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [debug] (ID 3) - Controller is executing TEST UNIT READY, CDB $00:00:00:00:00:00
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [debug] (ID:LUN 3:0) - Device is executing TEST UNIT READY ($00)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [trace] (ID 3) - Status phase, status is GOOD (status code $00)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [trace] (ID 3) - Sending 1 byte(s)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [trace] (ID 3) - All data transferred
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [trace] (ID 3) - MESSAGE IN phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.285] [trace] (ID 3) - Sending 1 byte(s)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.286] [trace] (ID 3) - All data transferred
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.286] [trace] (ID 3) - BUS FREE phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.289] [trace] (ID 3) - ++++ Starting processing for initiator ID 6
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.289] [trace] (ID 3) - SELECTION phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.289] [trace] (ID 3) - MESSAGE OUT phase
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.289] [trace] (ID 3) - Receiving 1 byte(s)
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.290] [trace] (ID 3) - Received IDENTIFY message for LUN 0
Oct 09 15:17:13 piscsi s2p[802]: [2024-10-09 15:17:13.290] [trace] (ID 3) - COMMAND phase
[... cut ...]
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.561] [trace] (ID 3) - Sending 1 byte(s)
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.561] [trace] (ID 3) - All data transferred
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.561] [trace] (ID 3) - MESSAGE IN phase
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.561] [trace] (ID 3) - Sending 1 byte(s)
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.562] [trace] (ID 3) - All data transferred
Oct 09 15:17:14 piscsi s2p[802]: [2024-10-09 15:17:13.562] [trace] (ID 3) - BUS FREE phase

```

uweseimet, 2024-10-09:
@Pacjunk Thank you. Looks as if the problem is related to a mode that is not implemented by SCSI2Pi:
```
Controller is executing READ(6)/GET MESSAGE(6), CDB $08:00:00:00:50:00
```
This command results in an error because SCSI2Pi does not support non-fixed tape blocks.  But your workstation also sends a MODE SELECT to configure something.
Can you try with a different block size, e.g. 1024 bytes ("-b 1024").
Please do not add more devices than required, it just makes the logfile more complicated. Just add what's absolutely needed, which I assume is a hard drive and a tape drive.

uweseimet, 2024-10-09:
@Pacjunk As far as I can tell the MODE SELECT tries to configure the density, setting it to the default value 0. 0 means that the device selects its default density, which is fine for SCSI2Pi, because other densities are not supported (or faked, because there is no density) anyway. The fact that your workstation eventually tries to read data indicates that it may be fine with whatever happened before (MODE SELECT/MODE SENSE).

The first actual error occurs when your machine tries to read a block of 80 bytes, which is currently not supported by SCSI2Pi. Supporting non-fixed block sizes is optional (and not required by tar). I will add this feature most likely tomorrow. Without this feature I don't expect a different behavior when configuring SCSI2Pi with a different blocksize ("-b 1024"), but I'd still like to ask you for doing this. It may provide further information. Please only use the debug log level, in order to keep the logfile small.

Do you have a real streamer that works correctly with your workstation? In this case you can run some SCSI commands against it (using s2pexec, the SCSI2Pi Wiki provides details about this tool). Knowing the mode pages data of your streamer might help to improve the emulation.

uweseimet, 2024-10-10:
@Pacjunk There is an issue_93_2 branch now, which is supposed to support variable block sizes between 1 and 4096 bytes. Can you please test whether something changes with this branch? But before doing that can you please test with the develop branch and a block size of 1024 bytes and comment in my questions in my previous comment? Thank you!

I cannot test flexible block sizes with the software I have, but I verified that tar still works with the issue_93_2 branch, and that tapeinfo reports the expected minimum and maximum sizes:
```
>tapeinfo -f /dev/sg5
Product Type: Tape Drive
Vendor ID: 'SCSI2Pi '
Product ID: 'SCSI TAPE       '
Revision: '0400'
Attached Changer API: No
MinBlock: 1
MaxBlock: 4096
SCSI ID: 2
SCSI LUN: 0
Ready: yes
BOP: yes
Block Position: 0
ActivePartition: 0
EarlyWarningSize: 0
NumPartitions: 0
MaxPartitions: 0
```

Pacjunk, 2024-10-10:
Thanks for that. I'll be away for a few days, so can't test until next week. I assume the first thing that the workstation does is look for the tape label. Yes I have some real tape drives - hopefully they and the tapes still work! I'm pretty sure that I dont get the device offline message with a blank tape - probably will just say label not found or not initialised. I'll have to try and bulk erase a tape and give it a try.

I'm using a pi zero W with bullseye. How do I get a copy of that branch? Is there a .deb file for it, or do I have to do a full build?

uweseimet, 2024-10-10:
@Pacjunk I have just triggered building .deb binaries for commit ID ffba888, which is the last commit in the issue_93_2 branch. The builds will be available in about an hour. The development binaries have the commit ID in their filename.

It would be very helpful if you could collect information on your real drives. You can do this by connecting your streamers to the PiSCSI board and running some SCSI commands. Just connect a streamer and then run commands with s2pexec, e.g.:
```
s2pexec>-i SCSI_ID_OF_YOUR_STREAMER -c 12:00:00:00:ff:00
00000000  01:80:02:02:1f:00:00:00:53:43:53:49:32:50:69:20  '........SCSI2Pi '
00000010  53:43:53:49:20:54:41:50:45:20:20:20:20:20:20:20  'SCSI TAPE       '
00000020  30:34:30:30                                      '0400'
s2pexec>-c 1a:00:3f:00:ff:00
00000000  47:00:00:08:00:1f:9f:dc:00:00:02:00:01:0a:26:01  'G.............&.'
00000010  00:00:00:00:01:00:00:da:02:0e:00:00:00:00:00:00  '................'
00000020  00:00:00:00:00:00:00:00:0a:06:00:00:00:00:00:00  '................'
00000030  10:0e:00:00:00:00:00:00:40:00:10:00:00:00:00:00  '........@.......'
00000040  11:06:00:00:90:00:00:00      
 s2pexec>-c 1a:00:00:00:ff:00
00000000  0b:00:00:08:00:00:00:00:00:00:02:00              '............'
```
This sequence of commands runs INQUIRY and and gets all mode pages with MODE SENSE and explicitly retrieves page 0.

Regarding s2pexec also see https://github.com/uweseimet/scsi2pi/wiki/Advanced-Testing.

bog-dan-ro, 2024-10-11:
@uweseimet 
I took a look at your code. Sadly I can not test it right now, my computer doesn't have RAM anymore[0].
A few things regarding your code:
 - you don't have specific code for `ModeSelect` & `ModeSense6|10`. I remember that I had to add specific implementation in order to make my computer happy. I'll double check once my computer is complete.
 - you're missing the `LoadUnload (1Bh)` command. I don't remember why I added it, but might be important.
 - I have more entries in `SetUpModePages`, I don't know if they are mandatory. Also the implementation of all AddXXXX methods are different.
 
Overall your code has looks better than mine, it has also proper unit tests!

[0] I temporary removed it and sent it to a friend to try to reverse engineering it. This computer uses custom RAM modules.
I'll test your implementation on ,my computer when it's complete, which might take a couple of weeks (I hope no more than a month).

uweseimet, 2024-10-11:
@bog-dan-ro You have probably missed that code for MODE SENSE/SELECT, LOAD/UNLOAD and mode page handling is inherited from the StorageDevice class.

uweseimet, 2024-10-11:
@Pacjunk The issue_93_2 branch has been merged into the develop branch.

Pacjunk, 2024-10-21:
Tried with block size set to 1024, got errors to do with no label (it does have a label).

Tried with your latest code (as of a couple of days ago), and get device errors. Debug log attached...

```
Oct 21 15:07:58 piscsi s2p[654]: +----+-----+------+-------------------------------------
Oct 21 15:07:58 piscsi s2p[654]: | ID | LUN | TYPE | IMAGE FILE
Oct 21 15:07:58 piscsi s2p[654]: +----+-----+------+-------------------------------------
Oct 21 15:07:58 piscsi s2p[654]: |  0 |   0 | SCHD | /home/pi/images/boot72.hd1
Oct 21 15:07:58 piscsi s2p[654]: |  3 |   0 | SCTP | /home/pi/images/vms_v5.4_1.tap
Oct 21 15:07:58 piscsi s2p[654]: +----+-----+------+-------------------------------------
Oct 21 15:08:36 piscsi s2p[654]: [2024-10-21 15:08:36.867] [info] Set log level to 'debug'
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.607] [debug] (ID 3) - Controller is executing TEST UNIT READY, CDB $00:00:00:00:00:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.607] [debug] (ID:LUN 3:0) - Device is executing TEST UNIT READY ($00)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.613] [debug] (ID 3) - Controller is executing INQUIRY, CDB $12:00:00:00:24:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.614] [debug] (ID:LUN 3:0) - Device is executing INQUIRY ($12)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.616] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.616] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.619] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.620] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.622] [debug] (ID 3) - Controller is executing MODE SELECT(6), CDB $15:00:00:00:0c:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.622] [debug] (ID:LUN 3:0) - Device is executing MODE SELECT(6) ($15)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.623] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.624] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.629] [debug] (ID 3) - Controller is executing READ POSITION, CDB $34:00:00:00:00:00:00:00:00:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.629] [debug] (ID:LUN 3:0) - Device is executing READ POSITION ($34)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.631] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:10:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.632] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.637] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:50:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.637] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.642] [debug] (ID 3) - Controller is executing REZERO/REWIND, CDB $01:00:00:00:00:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.642] [debug] (ID:LUN 3:0) - Device is executing REZERO/REWIND ($01)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.645] [debug] (ID 3) - Controller is executing TEST UNIT READY, CDB $00:00:00:00:00:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.645] [debug] (ID:LUN 3:0) - Device is executing TEST UNIT READY ($00)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.646] [debug] (ID 3) - Controller is executing INQUIRY, CDB $12:00:00:00:24:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.646] [debug] (ID:LUN 3:0) - Device is executing INQUIRY ($12)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.649] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.649] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.654] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.654] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.658] [debug] (ID 3) - Controller is executing MODE SELECT(6), CDB $15:00:00:00:0c:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.658] [debug] (ID:LUN 3:0) - Device is executing MODE SELECT(6) ($15)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.662] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:00:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.662] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.666] [debug] (ID 3) - Controller is executing READ POSITION, CDB $34:00:00:00:00:00:00:00:00:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.666] [debug] (ID:LUN 3:0) - Device is executing READ POSITION ($34)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.669] [debug] (ID 3) - Controller is executing MODE SENSE(6), CDB $1a:00:10:00:ff:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.669] [debug] (ID:LUN 3:0) - Device is executing MODE SENSE(6) ($1a)
[... cut ...]
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.720] [debug] (ID:LUN 3:0) - Device is executing REZERO/REWIND ($01)
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.724] [debug] (ID 3) - Controller is executing READ(6)/GET MESSAGE(6), CDB $08:00:00:00:50:00
Oct 21 15:09:00 piscsi s2p[654]: [2024-10-21 15:09:00.724] [debug] (ID:LUN 3:0) - Device is executing READ(6)/GET MESSAGE(6) ($08)

```

There is a SCSI test utility that tests various mode pages. I will run this and get back to you...

Pacjunk, 2024-10-21:
Here is the output from the SCSI test utility. I haven't used the scsi2pi tester yet as I need to either find some different cables or an external tape drive that works (I'm sure there is one somewhere!).

First file is with the SCSI2Pi running...

[SCSI2Pi.txt](https://github.com/user-attachments/files/17455472/SCSI2Pi.txt)

Next is connected to a Sony DAT drive. This is a generic drive and doesn't have any DEC specific functions.

[Sony DAT.txt](https://github.com/user-attachments/files/17455474/Sony.DAT.txt)

and finally, this is the command run again with a tape mounted. The information returned seems to be different.

[Sony DAT Mounted.txt](https://github.com/user-attachments/files/17455475/Sony.DAT.Mounted.txt)

Surprised the drive and especially the tape worked. Must be 20+ years old.

uweseimet, 2024-10-21:
@Pacjunk Thank you for running these tests.

There are some differences in the mode pages which are caused by SCSI2Pi supporting some pages the tape drive does not support. This is when " CHECK CONDITION seen" is reported for the tape drive, but no errors are reported for SCSI2Pi. These differences should not matter.
At the other data I will have a closer look at, especially in case of the medium partition pages. It looks as if your tape drive reports errors when returning data for partition numbers > 1, but SCSi2Pi does not, even though it should because only a single partition is supported.
In order to further analyze these data I would like to ask you for a SCSI2Pi log on trace level when you run your SCSI test utility, so that I can see the details of the commands and parameters sent by the tool.

In the SCSI2Pi log, is the " READ(6)/GET MESSAGE(6) ($08)" line the last line of the logfile, i.e. nothing happened after that? There are no SCSI errors logged anymore, but the sequence of MODE SELECT/MODE SENSE commands looks as if the workstation is trying to configure something. Can you please run the same test again, but with a block size of 4096 bytes and on trace level? Please use the very latest development binaries.
Depending on this logfile the next test would probably be to send the same MODE SELECT commands to the tape drive with s2pexec and then check what has changed with MODE SENSE.

> Tried with block size set to 1024, got errors to do with no label (it does have a label).
It does have a label? I'm afraid I don't know what you mean, assuming that a label is an object on a tape. A blank image file does not have any data on it, i.e. it would also have no label.

uweseimet, 2024-10-21:
@Pacjunk After running tests and collecting logs with the latest development binaries as outlined in my previous comment, I would like to ask you to run tests with the issue_93_3 branch. This branch you will have to compile yourself, and depending on future test results I will try to update it where required. In order to save compilation time I suggest using "make s2p", so that only the s2p binary is created. I expect that future updates will only affect the tape.cpp file, i.e. compiling updates will not take that long even on a small Pi.

In the issue_93_3 branch I modified the tape properties based on the mode page output of your tool, the mode pages of my HP streamer and information, and the SSC-5 specification, which is a bit more precise than the SCSI-2 specification. Nevertheless, there are some mode page fields/device properties where I do not know how to correctly support them. I hope that a SCSI2Pi trace log of the issue_93_3 branch provides more insight.

uweseimet, 2024-10-23:
After applying a work-around (not related to SCSI2Pi) I can now successfully use the Gemar tape backup software for the Atari. This software uses a fixed block size and works similar to tar.
On the one hand this means that besides tar there is another backup software that works with the current SCSI2Pi implementation, on the other hand this does not provide more insight.

uweseimet, 2024-10-28:
@Pacjunk Any news? I'm afraid that without feedback on my previous questions and more testing from your side it will not be possible to support more software/platforms than those that are currently supported: mt, mtx, tar and the Gemar streamer software for the Atari.

Pacjunk, 2024-10-28:
Sorry, life got in the way in a stressful and busy way. I will try to look soon. I assume I can just use the latest dev daily build?

uweseimet, 2024-10-28:
Yes, the latest daily build plus a self-compiled issue_93_3 branch, all with a block size of 4096 bytes, with log level trace. Also please see my previous comments.
It would also help to know the properties of all your physical drives, extracted by your tool.

Pacjunk, 2024-10-29:
Took forever to build the issue_93_3 branch, but got there in the end. Attached is the log. Note there is some access to device 0 at the beginning. This is just the OS scanning the SCSI buss. Still getting a device error when trying to mount a tape.

[29.txt](https://github.com/user-attachments/files/17553096/29.txt)

uweseimet, 2024-10-29:
@Pacjunk Thank you for the log. Please (in further logs) suppress logging for device 0: by using "-L trace:3". Also please remember to provide trace logs for the current develop branch, and to add the property data of your other streamer, when you find the time for doing so.

uweseimet, 2024-10-29:
I'm afraid there has been a misunderstanding. I would need logs on trace level for both branches. issue_93_3 differs from the develop branch. For the develop branch there are nightly binaries you can use. Please use the latest ones and provide the commit ID you use for testing.
Please confirm that the latest log is from the issue_93_3 branch. This branch and the develop branch have the same version number.

uweseimet, 2024-10-29:
The last log does not reveal any SCSI errors. If your computer shows an error it is somehow not receiving the data it expects.
Previously you mentioned something about a tape label. Please elaborate. Just like a blank tape, an image file does not contain any label. You have to explicitly create it, e.g. with a tape drive tool, regardless of whether there is a real drive or an emulated one.
The last SCSI command executed returns 80 bytes of data just as requested. If this is supposed to be a label, it's clear that your computer is not satisfied with the data received, because a label has never been created.
The bacula tape drive software for Linux requires each tape to be explicitly labeled before use. Looks as if you first have do that before you can use a tape image file.

uweseimet, 2024-10-29:
@Pacjunk After revisiting your latest log I wonder whether you were really testing the issue_93_3 branch. The last read command in the log, when executed for a blank image or an image with random data, should report either MEDIUM ERROR or BLANK CHECK. The code says so, and I have verified this behavior by sendin:
```
s2pexec>-i 3 -c 08:00:00:00:50:00 -L debug:3
[17:37:28.837] [debug] (ID 3) - Controller is executing READ(6)/GET MESSAGE(6), CDB $08:00:00:00:50:00
[17:37:28.837] [debug] (ID:LUN 3:0) - Device is executing READ(6)/GET MESSAGE(6) ($08)
Error: Can't execute command READ(6)/GET MESSAGE(6) ($08)
s2pexec>[17:37:32.650] [debug] (ID 3) - MEDIUM ERROR (Sense Key $03), READ ERROR (ASC $11), ASCQ $00
```
The error should be the same with the develop and the issue_93_3 branch.

In order to avoid any potential confusion and mismatches between the code I am working on and the code you are testing, please:

- Ensure that you are using an empty (fresh) tape image file, created by dd or fallocate.
- For testing, do not run the s2p system service, but launch s2p from the command line, e.g. with /opt/scsi2pi/bin/s2p when testing a binary distribution installed with apt, or with sudo ./cpp/bin/s2p when executing s2p from a self-compiled source tree.
- In general, when running s2p as a system service do not use a modifed piscsi system service file (the logfile says piscsi, that's why I assume it is not the original s2p service).
- Always provide the respective commit ID with your logs, in the ideal case as part of the logfile filename.

Thank you!

uweseimet, 2024-11-01:
@Pacjunk You mentioned issues with the PiSCSI web UI and SCSI2Pi. While waiting for your feedback I verified whether the web UI is working with SCSI2Pi and did not notice any problems. I installed 
2024-04-30-PiSCSI-v24.04.01-arm64-lite.zip, and then I installed SCSI2Pi 3.4.3 with apt. After starting the s2p system service (see Quickstart on https://www.scsi2pi.net/en/installation.html) I could properly access the UI, see the attached screenshot.
Note that the web UI does not work anymore after updating the Pi to the latest (bullseye) packages before and maybe also after) launching the web UI the first time. Looks as if the current bullseye setup of pip-managed python packages is not compatible with what the web UI requires. But this is a PiSCSI web UI issue and not caused by SCSI2Pi.
[image]

Pacjunk, 2024-11-06:
Hi, got a couple of hours today to do some testing using 13d6667 (although same results with 5cb8b4f). Using a blank tap file I managed to initialise (ie write a label), then mount the tape with that label successfully. Trying to copy files to the image resulted in errors. I notice that every record written to the .tap file has a "SCTP" prefix. This is most likely the reason that I cannot mount .tap files produced by others from places such as archive.org. Of course they don't have records prefixed with SCTP!

I have attached two files. The .tap file (after it has been initialised) and the log file. The steps were:
1 - Initialise the tape with a label
2 - Mount the tape
3 - Try to copy a file to the tape (failed)
4 - Dismount the tape.

I probably won't get any time with this until next week as I will be away for a few days. Cheers,

[2.log](https://github.com/user-attachments/files/17643678/2.log)
[2.tap.txt](https://github.com/user-attachments/files/17643690/2.tap.txt)

uweseimet, 2024-11-06:
@Pacjunk Thank you for the logs. Yes, the non-tar format is proprietary because it also contains meta data (see the comment at the beginning of tape.cpp) that represent filemarks and end-of-data markers. Without these meta data spacing in order to search for a filemark, for example, would not be possible. This is why spacing cannot be supported with tar files: The required markers are missing in these files.

If I understand your correctly the log does *not* contain the SCSI operations executed when labeling a tape?

Pacjunk, 2024-11-06:
Yes the log includes labelling the tape.

Pacjunk, 2024-11-10:
> Is there any information on the format of these files? Assuming that these are not just tar files, they must contain encoded meta data that represent filemarks etc., and this encoding must be document somewhere in order to be usable. From your latest log it is obvious that filemarks are used to mark labels. Any kind of encoding I can think of would require rather ugly code to read/write it, and/or would increase the file size quite a lot. Especially when you want to space backwards, you need to somehow find where the previous tape object (block, filemark etc.) starts. This information cannot be encoded in the actual block data, but must live in a separate world. I guess that with a real streamer there is an interrupt when such kind of data are encountered, and the streamer's firmware can then work with these data, without confusing them with an actual block payload. IMO the description of the SPACE(6) command and what happens when a block/filemark/setmark/end-of-data marker is encountered unexpectedly implies this.
> 
I don't know how they are structured. I do know that simh reads them correctly. Whether the .tap file is specially formatted for simh, or simh follows some standard, I don't know.

uweseimet, 2024-11-10:
@Pacjunk Do you have a download link for such a file?

uweseimet, 2024-11-10:
@Pacjunk Regarding reverse spacing I would like to verify whether your drive actually supports it. Can you please connect your drive to the PiSCSI board and then launch s2pexec and send the following commands to your drive:
```
s2pexec>-i SCSI_ID_OF_YOUR_DRIVE -L debug -c 12:00:00:00:ff:00
s2pexec>-c 10:00:00:00:01:00
```
If for any command "Error: UNIT ATTENTION" is reported (probably for the second command), please send the same command again.
After each command there will be some console output, which I am interested in. The command do not modify any data.

Pacjunk, 2024-11-11:
Here are some files...   http://www.bitsavers.org/bits/DEC/vax/vms/TK50/

I was doing some reading, and the labelling conforms to the ANSI tape label format.

Wouldn't it be up to the OS (or driver) to look after tape marks etc, rather than the tape drive inserting them?

I will have to hunt around for appropriate cables to connect the tape drive and piscsi board up simultaneously.

uweseimet, 2024-11-11:
@Pacjunk I see. The goal of SCSI2Pi is to support regular tar files and a proprietary image file format that reflects most of the objects that SCSI commands can write to a tape, using the native tape driver of a real machine. This is a different approach than supporting simh files.
Looks as if there are these 3 kinds of relevant formats, all with their special use case:

1. tar archive files. They do not contain markers like filemarks, just the plain data created by tar and compatible tools. On Linux "tar cf /dev/st0 FOLDER_TO_ARCHIVE" creates them, "tar xf /dev/st0" extracts them.
2. simh files, which are platform independent but require special drivers for reading/writing them.
3. raw image files, which work with the native drivers for a platform. They are platform-specific because the respective OS/driver determines the format and decides on the use of filemarks etc.

Do you agree with this list? SCSI2Pi aims at supporting items 1. and 3. Items 1. and 3. have already successfully been tested with tar on Linux and Gemar on the Atari, but tar and Gemar do not make use of tape drive features like spacing, in particular reverse spacing, which you need for your platform.
Supporting item 2. is probably interesting, but it requires a documentation on the simh format at least.

Pacjunk, 2024-11-11:
Yep, fine. I didn't expect SCSI2Pi to support simh files. I just used them as a ready-made input file for testing. Looking at the spec it seems that the record length is specified at the beginning and end of the data block to enable reversing. Looks like this sort of behavior will be too complicated for this project (although I assume real physical drives must do it - need to do those tests!)

Spec is here: http://simh.trailing-edge.com/docs/simh_magtape.pdf

uweseimet, 2024-11-11:
@Pacjunk Yes, adding the length or previous position also at the end (currently SCSI2Pi only adds it to the beginning) of an object (I use the term object because the specification also uses it, it covers not just data blocks but also filemarks etc.) I something I have already thought about. But I have not made up my mind yet.
I am also not 100% sure whether your driver in normal operation really needs reverse spacing support. It might be that the driver just tries reverse spacing as a last resort because it did not find what it was looking for on the tape. This is why I asked for checking whether your drive really supports reverse spacing. My drive appears to support it, but as already mentioned it is broken and cannot write data anymore. But it does not reject the SPACE command, i.e. it knows the command.

Pacjunk, 2024-11-11:
The document at one point talks about physical drives reversing in order to re-read bad blocks. It specifically mentions TK50 which was a common software distribution media before CD-ROMs came along.

I will try to test one (maybe two) of my tape drives in the next couple of days.

uweseimet, 2024-11-11:
@Pacjunk A bad block is indeed something I was thinking of as a reason for reverse spacing. This is what I meant with "in normal operation": There cannot be bad blocks in our case, and if the driver thinks that there is a bad block, something must have gone wrong before. But finding out what might have gone wrong is hard. When analyzing the logfile I did not find anything that was obviously wrong or looked like something the driver could regard as a bad block. My guess is that there is still something wrong with the filemarks handling. The rules of how spacing works, how objects are skipped during spacing, and how spacing results are returned in the sense data are quite complicated.
The latest issue_93_3 branch contains some improvements regarding the sense data returned when spacing. Can you please run a new test? Please try to limit the amount of operations to a minimum, e.g. when labelling do it exactly once. Note that the raw file format has changed. Do not reuse any old .tap files but please use new, empty image files. Also note that the default extension has changed to .st. If you want to also use .tap as an extension that is automatically recognized add this line to /etc/s2p.conf:
```
extensions.sctp=tap
```

uweseimet, 2024-11-11:
After having a superficial look at the simh format, this format probably provides everything I need for item 2 on my list, including gap support. This would be nice, because I prefer existing formats over proprietary ones.

I will create a separate ticket for raw tape files support (be it in simh format or something proprietary). The initial focus of tape support in SCSI2Pi 4.0, which is also mentioned on the SCSI2Pi website, is support for image files that can be used with tar (and as it turned out also with Gemar). This feature has already been successfully tested, and therefore it makes sense to have a release 4.0 that comprises this feature. simh (or similar) support will then become part of release 4.1 or so.

A new ticket will also make it easier to follow any progress, without having to read all comments in the current ticket.

@Pacjunk As far as testing is concerned, nothing substantial is going to change, because anything that might be wrong with the spacing implementation will also be wrong with a different file format.
This means that it would still be useful if you could run a test (trace level) with the current issue_95_3 branch, and to connect a real tape drive to the board in order to verify whether the drive supports reverse spacing.

uweseimet, 2024-11-11:
https://github.com/uweseimet/scsi2pi/issues/100 is the follow-up ticket to https://github.com/uweseimet/scsi2pi/issues/93.

uweseimet, 2024-11-11:
SCSI2Pi 4.0 release candidate binaries with streamer support for tar files only are now available. There is already an issue_100 branch for https://github.com/uweseimet/scsi2pi/issues/100. I will need some time to change the existing code to read/write the simh format. In issue_100 the .tap extension is supported again by default, this time with the goal to be simh-compatible, which makes this extension the correct choice.

@Pacjunk Do you think I would profit from installing simh on my Linux machine? I assume that I would not be able to use SCSI2Pi in a simh emulation context, even with a Linux-PC with SCSI card. But maybe I am missing something and there is a setup that would help with testing.

closed 2024-11-11 by uweseimet

ppuskari, 2024-11-11:
I am finally able to get some time on this project.  I hat I am fallen behind on is how to setup for testing.    I have Piscsi device and Linux and compilers etc per not mail to download Piscsi sources and compile and to compile my custom netatalk and Ghostscript ImageWriter II color support etc.

What links do I need to also grab or what fork to get the latest tape pieces as well as the test apps we are using in this thread.  I can get setup and test that Apple 40meg tape support 3m block addressable tape after a full tape format.

Things look promising.  And many thanks guys for working on this!

Petar Puskarich
***@***.***

Closed #93<https://github.com/uweseimet/scsi2pi/issues/93> as completed.