uweseimet/scsi2pi #111: Test tape device with a VAXstation 3100 running OpenVMS 7.2

uweseimet, 2024-12-04:
This is a follow-up ticket for https://github.com/uweseimet/scsi2pi/issues/100.

uweseimet, 2024-12-04:
These are the logs and scripts for this ticket.

- Foreign mount script:
[issue_111_script.txt](https://github.com/user-attachments/files/18007048/issue_111_script.txt)

- Logfile when sending this script to s2p:
[mount_s2p.txt](https://github.com/user-attachments/files/18007060/mount_s2p.txt)

- Logfile when sending this script to the Sony tape drive:
[mount.txt](https://github.com/user-attachments/files/18007003/mount.txt)

@pacjunk Please add whether may be missing.

Pacjunk, 2024-12-04:
Correction: The mount/foreign works correctly. The error occurs when using the backup/list command to list the contents of the first saveset.  If the tape is mounted normally (files11), then the first file can be copied from tape to disk and read successfully - which shows that the tape is formatted correctly, and the files are valid. Simh can read the same .tap file (although not emulating SCSI), and a physical SCSI tape drive also works correctly.

uweseimet, 2024-12-04:
I sse, I confused the commands. s2p can also read the file, but it has to report a SCSI error because of the block mismatch. s2p does not have an issue with the file, but with the combination of block sizes in the file and the SCSI command to read these blocks.
Can you please attach the s2p log that results in your OS complaining?

Pacjunk, 2024-12-04:
This is the trace log for the mount/foreign command - which completes successfully.
[mount.txt](https://github.com/user-attachments/files/18007353/mount.txt)

This is the log for the backup/list (what I was calling foreignlist) - which fails and produces the error "tape is not valid ANSI format". This is because it attempts to read the label block and s2p signals an error because of block size mismatch.

[list.txt](https://github.com/user-attachments/files/18007381/list.txt)

uweseimet, 2024-12-04:
Just to ensure that the test scenario for this ticket is correct: First mount.txt has to be applied, then list.txt. With your drive this sequence of commands works fine, with s2p it results in an error.

Pacjunk, 2024-12-04:
Correct

uweseimet, 2024-12-04:
Good, but these were only the logs. Can you please add the matching scripts?

uweseimet, 2024-12-04:
A question regarding the ANSI format: The initial blocks in the .tap file we use for testing is 80 bytes. Is this ANSI format? I am asking because your OS reports an ANSI-related error when trying to read 90 byte blocks from this file. If the 80 bytes are ANSI format, trying to read 90 bytes can never succeed, also not with a real tape drive.

Pacjunk, 2024-12-04:
Yes, that is ANSI.

I found some information on VMS 8.4 (quite a bit later, and not using this architecture, but tape formats havent changed for years). Link is https://community.hpe.com/t5/operating-system-openvms/openvms-8-4-backup-log-disk-test-mga0-test-bck-save-not-working/td-p/4830493. The person is getting the same error messages as me. One of the replies states the following:

`backup sets up a 90-byte buffer for reading the VOL1 header and EXPLICITLY checks the IO status block for 80 bytes being returned from the QIO and returns BACKUP$_NOTANSI, if that check fails.
`
It contains the magic number "90". I know this is not a proper solution, but what would happen if when the request is for 90 bytes, s2p just returns the data in the block (80)?

uweseimet, 2024-12-04:
Where would take s2p the additional  data from? Note that your tape drive also does not return more data than available, but signals the mismatch in the response, in the same way s2p does. RQUEST SENSE then provides the details, with custom bytes at the end of the added by your drive.

Pacjunk, 2024-12-04:
What additional data? The above statement says that the device just returns the 80 bytes (and I assume there is a length field somewhere indicating the length of the response?)

The tape drive must at some stage return the required data though, otherwise it wouldn't work.

uweseimet, 2024-12-04:
The drive in the sense data provides the difference of requested vs. available data, and that's it. It will never return more data than the block size, because more data for the requested block just do not exist. We have already verified that your drive and s2p handle this case identically, and according to the specs.

IMO the fact that your OS tries to read 90 bytes at all is something that should not happen. I bet with your drive it does not happen, because we have already seen that the drive would report the same error as s2p, except for the custom bytes.
If the custom bytes have no relevance, it means that something must have happened before, which caused your OS to do what is does in the s2p case.
I need to check all commands in the logs again, but I would need current logs for the two relevant operations.

Pacjunk, 2024-12-05:
With the mode page stuff set, the only difference is the mount now reports that the tape is write locked.

[mount.txt](https://github.com/user-attachments/files/18020997/mount.txt)
[list.txt](https://github.com/user-attachments/files/18021001/list.txt)

I will run the script soon...

uweseimet, 2024-12-05:
OK, thanks a lot.

I may just have found something relevant while re-checking all logs. I still don't know why my streamer does not work with Linux and the Atari. When using s2pexec with it I can successfully run the init script and can read the data it writes. When doing this I noticed that even that in the mount log there is a difference in the error data returned when the first filemark is hit. The OS reads the first 3 blocks, then there is another read command. This one has to fail because after the third block there is a filemark. This raises an error (with your drive, with my drive, and with s2p), but s2p does *not* return the actual block size in the error message, whereas my drive does. I am in the process of updating s2p accordingly.

Pacjunk, 2024-12-05:
I ran the script twice. The first time it produced a message "Error: UNIT ATTENTION (Sense Key $06), NOT READY TO READY TRANSITION (MEDIUM MAY HAVE CHANGED) (ASC $28), ASCQ $00". The second time, no such message.

[output.txt](https://github.com/user-attachments/files/18021170/output.txt)
[output2.txt](https://github.com/user-attachments/files/18021172/output2.txt)

uweseimet, 2024-12-05:
Wrt your latest comment: I don't know what you mean with "mode page stuff set".

uweseimet, 2024-12-05:
Running the script on your drive has revealed what the MODE SELECT does. With the MODE SENSE before the MODE SELECT your drive says that it has a default block size for fixed size read commands of 512 bytes. The MODE SELECT changes this. After the MODE SELECT your drive return 0 as default block size. The spec says:
`"The block length specifies the length in bytes of each logical block
described by the block descriptor.  For sequential-access devices, a block
length of zero indicates that the logical block size written to the medium
is specified by the transfer length field in the command descriptor block
(see 10.2.4 and 10.2.14)"`
This is not yet what s2p does. I will change this.

uweseimet, 2024-12-05:
The block size change works now. Can you please check if this already improves something after updating and re-compiling issue_100? Important: Remove the mode page line from /s2p.conf.

Pacjunk, 2024-12-05:
The mount crashes s2p with floating point exception

[mount.txt](https://github.com/user-attachments/files/18022319/mount.txt)

uweseimet, 2024-12-05:
I just committed a fix so that when htting a filemark the requested length is returned. The exception is most likely also fixed in the very latest issue_111 sources.

Pacjunk, 2024-12-05:
Same floating point exception

[mount.txt](https://github.com/user-attachments/files/18022790/mount.txt)

uweseimet, 2024-12-05:
I see, please provide the script, so that I can reproduce the issue with my setup.

uweseimet, 2024-12-05:
I may have found what's wrong without a script (but would still like to have it). Please update once more and re-test.

Pacjunk, 2024-12-05:
That fixed the floating point exception. Logs for mount:
[mount.txt](https://github.com/user-attachments/files/18023071/mount.txt)
[mount_script.txt](https://github.com/user-attachments/files/18023072/mount_script.txt)

Backup/list still fails. Logs:
[list.txt](https://github.com/user-attachments/files/18023076/list.txt)
[list script.txt](https://github.com/user-attachments/files/18023077/list.script.txt)

uweseimet, 2024-12-05:
Tthe sense data when hitting the filemark in the mount test are now correct.

Please configure a custom mode page 0 again, this time with a block size 0, by updating /etc/s2p.conf:
```
mode_page.0.SCSI2Pi:SCSI TAPE=0b:00:90:08:13:00:00:00:00:00:00:00
```
Then please run the list test and provide script and log.

Pacjunk, 2024-12-05:
[list.txt](https://github.com/user-attachments/files/18023250/list.txt)
[list script.txt](https://github.com/user-attachments/files/18023255/list.script.txt)

I'm packing up for the night. I will check in tomorrow. Cheers,

uweseimet, 2024-12-05:
Thank you, and good night. At least two more issues have been fixed by comparing with real drive data. I will later evaluate the new logs.

uweseimet, 2024-12-05:
I updated s2pexec to log all we need on debug level, without all the noisy bus phase etc. messages, which are now only logged on trace level. From now on, please add "-L debug" at the beginning of each script you execute with s2pexec. This will make the logs much shorter and better readable.

Pacjunk, 2024-12-06:
Where is the init script, or do you just want me to label the tape with the OS?

uweseimet, 2024-12-06:
Yes, that's what I meant, i.e. running the process that creates a file of this structure:
```
Offset 0 ($0): Class 0, good data record, record length 80 ($50)
Offset 88 ($58): Class 0, good data record, record length 80 ($50)
Offset 176 ($b0): Class 0, good data record, record length 80 ($50)
Offset 264 ($108): Class 0, tape mark
Offset 268 ($10c): Class 0, tape mark
Offset 272 ($110): Class 0, good data record, record length 80 ($50)
Offset 360 ($168): Class 0, good data record, record length 80 ($50)
Offset 448 ($1c0): Class 0, tape mark
Offset 452 ($1c4): Class 0, tape mark
Offset 456 ($1c8): Class 7, private marker (SCSI2Pi end-of-data object)
```
I use this script for this because I cannot have any OS do this:

[init_script.txt](https://github.com/user-attachments/files/18034973/init_script.txt)

We have to ensure that regardless of with or without OS, with or without real tape drive, our test scenario is always equivalent with running init, mount and list in this order.

Pacjunk, 2024-12-06:
OK here is what I did...

I ran your init script, then the mount script, then the list script:
[init_out1.txt](https://github.com/user-attachments/files/18035331/init_out1.txt)
[mount_out1.txt](https://github.com/user-attachments/files/18035332/mount_out1.txt)
[list_out1.txt](https://github.com/user-attachments/files/18035336/list_out1.txt)

Then I attempted to mount the tape in the OS that had been initialised with your script. This worked for both Files11 and foreign mounts.

I then re-initialised the tape with the OS (same label) and reran the mount and list scripts:
[mount_out2.txt](https://github.com/user-attachments/files/18035363/mount_out2.txt)
[list_out2.txt](https://github.com/user-attachments/files/18035365/list_out2.txt)

uweseimet, 2024-12-06:
This means all in all, the init script, which is the only script that *writes* and data, does excactly do the same as labeling does. Correct me if I am wrong, please. The other two scripts only read, but do not write.

Pacjunk, 2024-12-06:
I just tested the mount command. To be sure I need to copy some files to and from the tape. I have switched the gear off for a little while as there is a thunderstorm rolling through. I'll test after that.

uweseimet, 2024-12-06:
Sure, no problem. The plan is to return exactly the same data as your drive. Even the synchronous transfer bit, which cannot be supported. When your OS sees exactly the same data, something must change in what the OS does when s2p is connected, because it cannot tell the difference between emulation and real hardware.

Pacjunk, 2024-12-06:
Somehow my tape got corrupted. I got errors when copying files to it.  I did an erase on the tape, then initialised it again and things work properly. I then re-ran your init script and copied files and this time it all worked.

I don't know what happened before and why it got corrupted. but seems OK now.

Pacjunk, 2024-12-06:
This is what happens when the OS boots. I assume it is just scanning the SCSI busses looking for devices.

[boot.txt](https://github.com/user-attachments/files/18037876/boot.txt)
[boot_script.txt](https://github.com/user-attachments/files/18037877/boot_script.txt)

Anyway, I'm off to bed...

uweseimet, 2024-12-06:
Thank you, my job is now to once again have a close look at everything.

uweseimet, 2024-12-06:
Result of comparing the two mount_out1 and mount_out2. There are differences, but in practice they do not matter:
1. When running mount_out2, the tape had not been rewound, according to what READ POSITION report. But your OS sends a REWIND anyway, and the subsequent commands were executed at the correct position, i.e. the beginning of the tape.
2. The "HDR1" labels have different numbers in them: "4341024329" and "4041024341". I guess these are timestimps and they do not matter. Maybe you know more about the meaning of these numbers?
Assuming that these numbers do not matter the logs show that the mount operation does the same with the tapes labeled by our OS as with the tapes labeled by the init script.

Pacjunk, 2024-12-07:
Build f5d15c2

This is a combined mount and list log (since you said temporary values may change if I stop s2p):

[mount&list.txt](https://github.com/user-attachments/files/18047634/mount.list.txt)
[mount&list script.txt](https://github.com/user-attachments/files/18047635/mount.list.script.txt)

I think it has got to the stage where we need to run scsimon or something similar to see what the real tape drive does about the 90 byte request. I have scsimon dumps, but don't know how to interpret them.

Pacjunk, 2024-12-07:
Yes, but the operation on the real drive is successful. So either it must ask for 90 bytes and get 80, or it must ask for 90, get rejected then ask for 80. Whatever it does, the operation succeeds. Thats why I think it would be useful to monitor the traffic to see how it works.

uweseimet, 2024-12-07:
I'm afraid I have to disagree: The read operation of 90 bytes is not successful with the real drive. s2pexec clearly shows that. And the specification says that a drive must *not* return data in this case.
This is why I doubt that the sequence of commands when your drive is connected results in this read operation at all. Something else must be different before.

uweseimet, 2024-12-07:
By the way, using a different drive would be a good idea at this point. I expect the same result, i.e. that it will just work, but maybe we learn something knew.

Pacjunk, 2024-12-07:
So if you believe that the OS never asks for a 90 byte block with the real tape drive, then it would be useful to monitor it and prove that.

uweseimet, 2024-12-07:
I would not know how to prove that. But I believe that this read never happens because we have proven by running such a command against your drive, that your drive will raise an error. And we know that the specification says that your drive has to do just that. This is consistent. This is why I think it is reasonable to believe that this read never happens.

uweseimet, 2024-12-07:
I am just wondering: Would using the simh emulator help? I don't have the required knowledge to use it, but I guess it is running a similar or the same OS in the emulation? The emulation must have its own drivers to access simh image files, but the OS would still run the same commands it runs on your real hardware, wouldn't it? And their might be means to monitor what the OS when running on simh does.

uweseimet, 2024-12-07:
What I am also wondering. Let's forget the technical details in the logs for a moment: Why would your OS think that this tape is a different kind of tape, i.e. one with 90 byte blocks? You have mentioned that this kind of blocks exists in a different context, haven't you? Why would the OS think that it is in such a context?

Pacjunk, 2024-12-07:
The tape drives in simh are not SCSI, so it would be hard to compare.

I can't see why the OS would look for actual 90 byte blocks, because it would then not be ANSI compliant.

uweseimet, 2024-12-07:
OK, so simh would not help. I had a brief look at the simh project pages yesterday. But since I have never used VMS I don't think that I would be the right person to try out simh.

Regarding the OS: From the logs we know that there are situations in which the OS does look for a 90 byte block, because it explicitly tries to read a block of this size. This means it knows something about non ANSI compliant blocks. When being connected to s2p it thinks that using this block size is a good idea. This is why I wonder when the OS might think that.

uweseimet, 2024-12-07:
I think we are at a point where it is really hard for the two of us to proceed. There were one or two other users who commented in the previous tickets I think. It would be good to have more testers, with other platforms. This might reveal something new, because using more/other drivers will provide additional insight.

Pacjunk, 2024-12-07:
simh does more than just VMS, but they are mostly minicomputer type systems with heaving leaning  towards DEC gear. Written by a DEC guy, so I suppose this makes sense. I think the tape drive stuff emulates a TQK50 controller. Certainly the device name (MU) supports that. If it was SCSI it would be MK.  There is a fork of simh that supports later workstations (including mine), but it only supports SCSI disks. Tape is still non-SCSI.

What would happen (as a test) if the s2p code didn't generate an error and just returned the 80 bytes? Would be interesting to see.

uweseimet, 2024-12-07:
Your OS requests 90 bytes in the DATA OUT phase. Just returning 80 will cause issues, e.g. a blocked bus. We simply have to accept that also the real drives do not return 80 bytes in this case, and that this is compliant with the specification. The drives and s2p do the right thing *in this context*.

Pacjunk, 2024-12-07:
Looks like we're never going to find out then!

If this is a dead end, I suggest we move onto writing files to the tape (reading seems to work)

uweseimet, 2024-12-07:
@bog-dan-ro You mentioned that you wanted to implement tape support for PiSCSI. This appears to be stalling, but anyway: I would appreciate your help with testing the SCSI2Pi tape support. The current state is already far beyond what your suggested code for PiSCSI provides. It also supports non-fixed block sizes and simh-compatible image files.
@Pacjunk and I have spent a lot of time with testing s2p against OpenVMS, but we appear to be in a dead end. This is why an additional tester, with an additional platform, would be very helpful. Maybe you would like to help us? In this case, please note that I would need rather prompt feedback on a regular basis.

uweseimet, 2024-12-07:
@Pacjunk I agree, let's continue with writing. I would not be surprised if we learn something when dealing with other features, i.e. writing, that helps us with the current problem.

uweseimet, 2024-12-07:
@sidick @ppuskari You also expressed interest in tape support. If you are still interested, this is the right time to help with testing.

uweseimet, 2024-12-07:
@Pacjunk I assume that you are going to start with a new test case. If there is something that does not work, please post the required data - you have a lot of practice with this now ;-) - and I might create a new, specific ticket for this case.

Pacjunk, 2024-12-07:
OK. Here are some logs. I initialised a tape in the normal way (don't think we need logs for that now).  The .tap file is 64KB in size.

I then mounted the tape (not foreign like we were doing). This worked.
[mount.txt](https://github.com/user-attachments/files/18047998/mount.txt)
[mount script.txt](https://github.com/user-attachments/files/18048001/mount.script.txt)

Then I tried to copy a small file (probably on 40 bytes of real data) to the tape...

[copy.txt](https://github.com/user-attachments/files/18048004/copy.txt)
[copy script.txt](https://github.com/user-attachments/files/18048005/copy.script.txt)

The main error message I got, said that "magnetic tape position lost". I assume it didn't find the right place to append the ANSI header and/or the data for the file.

uweseimet, 2024-12-07:
@Pacjunk Thank you. I will investigate this. I created https://github.com/uweseimet/scsi2pi/issues/112 for dealing with writing data to a tape with your platform. Please from now on let's use the new ticket for anything related to writing data, so that the current ticket remains focused on the 90 bytes etc. issue when reading.

uweseimet, 2024-12-07:
@Pacjunk Regarding the list operation, let me make some claims/assumptions. Please comment on what you think is wrong and why. This is not so much about s2p, but about the operations executed by the OS.

1. Your OS knows about 90 byte blocks, even though they may not be ANSI. Otherwise it would not try at all to read exactly 90 bytes. The 90 is not a random number.
2. Tapes that do not have 80 byte blocks at the beginning cannot be mounted.
3. Mounting a tape is a precondition for the list operation.
4. Because the tape could be mounted, the OS knows that the first block has 80 bytes.
5. After mounting a tape the tape is positioned at the beginning, because the mount operation executes a rewind at the end.
6. Despite this knowledge the OS tries to read 90 bytes from the beginning with the list operation

ppuskari, 2024-12-07:
Yes getting back into this and will get stuff deployed asap. Job change has really eaten into my play time. lol.

Appreciate all the work that you have done.

[128 more replies]

closed 2024-12-23 by uweseimet