uweseimet/scsi2pi #112: Test writing data to a tape with a VAXstation 3100 running OpenVMS 7.2

uweseimet, 2024-12-07:
This is a follow-up ticket to https://github.com/uweseimet/scsi2pi/issues/111, dealing with writing data.

uweseimet, 2024-12-07:
@pacjunk There is a new branch issue_112, which is based on the issue_111 branch. Any changes dealing with writing data (your copy script, for instance) will be committed to this new branch.

Pacjunk, 2024-12-07:
I included the mount script for your information, as the mount is done in a different way to our previous tests.

uweseimet, 2024-12-07:
Your copy script uses spacing and reverse-spacing, most likely in order to find the correct position for writing. I will review the respective code and will also try to run the script against my real tape. I will get back to you when I have committed something.
For subsequent tests, please remove the "device.3:0.name" property from /etc/s2p conf.

OK; I will also have a look at the mount operation.

uweseimet, 2024-12-07:
Based on your log and with the help of my streamer and the specification I fixed a bug in the object count the SPACE command returned. Please test again with the latest issue_112 branch.

Pacjunk, 2024-12-08:
Build 2101769

I have tried to copy again. This time the OS says the file was copied, but could not be closed. Looking at the .tap file, the text in the file that I copied is present, but it appears to have corrupted the file.
[copy.txt](https://github.com/user-attachments/files/18051129/copy.txt)
[copy script.txt](https://github.com/user-attachments/files/18051130/copy.script.txt)

As to your question, no this copy is different to what we were doing before. When the tape is mounted "normally", you can copy files to/from the tape, do directory commands etc just like the tape was a file system (except no directories!).

Mounting the tape "foreign" is used by the backup utility which creates save-sets (tar in unix speak). To list the **contents** of this  file, you do a backup/list command which is what is asking for the 90 byte read.

These tape operations should not do the 90 byte read as I am not using the backup command. Copying files from the emulated tape works fine. We are not trying to get it to write to the tape without corrupting it.

uweseimet, 2024-12-08:
The corrupt data are the only thing that is wrong, correct? Your OS thinks everything is fine? Can you please try the same with -b 8192? I guess there is a similar issue like their was with reading before.

Pacjunk, 2024-12-08:
The structure of the file seems incorrect. If you re-mount it in the OS, it appears blank, but you get different errors when trying to copy to it. -b 8192 makes no difference. There doesn't appear to be any end of file markers.

uweseimet, 2024-12-08:
Thank you. How big is the file? Maybe you can attach it, or can provide about 20 lines of s2psimh output for this file? Just the structure, no data, i.e. *not* with the -d option.

Pacjunk, 2024-12-08:
The filename seems to be in the 4th block. Other files downloaded from bitsavers, the first filename is in the second block.

File attached: 
[1.tap.dmp](https://github.com/user-attachments/files/18051720/1.tap.dmp)

uweseimet, 2024-12-08:
Can you please also add the s2psimh output for this file?

On first sight the data as such seem to be fine. I agree that the write position appears to be wrong. Your OS is using SPACE (for positioning) a lot.

This is a file after labeling:
```
Offset 0 ($0): Class 0, good data record, record length 80 ($50)
Offset 88 ($58): Class 0, good data record, record length 80 ($50)
Offset 176 ($b0): Class 0, good data record, record length 80 ($50)
Offset 264 ($108): Class 0, tape mark
Offset 268 ($10c): Class 0, tape mark
Offset 272 ($110): Class 0, good data record, record length 80 ($50)
Offset 360 ($168): Class 0, good data record, record length 80 ($50)
Offset 448 ($1c0): Class 0, tape mark
Offset 452 ($1c4): Class 0, tape mark
Offset 456 ($1c8): Class 7, private marker (SCSI2Pi end-of-data object)
```
This is the result of your test:
```
Offset 0 ($0): Class 0, good data record, record length 80 ($50)
Offset 88 ($58): Class 0, good data record, record length 80 ($50)
Offset 176 ($b0): Class 0, good data record, record length 80 ($50)
Offset 264 ($108): Class 0, tape mark
Offset 268 ($10c): Class 0, good data record, record length 80 ($50)
Offset 356 ($164): Class 0, good data record, record length 80 ($50)
Offset 444 ($1bc): Class 0, good data record, record length 80 ($50)
Offset 532 ($214): Class 0, good data record, record length 80 ($50)
Offset 620 ($26c): Class 0, tape mark
Offset 624 ($270): Class 0, good data record, record length 2048 ($800)
Offset 2680 ($a78): Class 7, private marker (SCSI2Pi end-of-data object)
```
The end-of-data marker is present. Correct me if I am wrong: The bitsavers files save the filenames at offset 88? This is the second block.

Pacjunk, 2024-12-08:
```
Offset 0 ($0): Class 0, good data record, record length 80 ($50)
Offset 88 ($58): Class 0, good data record, record length 80 ($50)
Offset 176 ($b0): Class 0, good data record, record length 80 ($50)
Offset 264 ($108): Class 0, tape mark
Offset 268 ($10c): Class 0, good data record, record length 80 ($50)
Offset 356 ($164): Class 0, good data record, record length 80 ($50)
Offset 444 ($1bc): Class 0, good data record, record length 80 ($50)
Offset 532 ($214): Class 0, good data record, record length 80 ($50)
Offset 620 ($26c): Class 0, tape mark
Offset 624 ($270): Class 0, good data record, record length 2048 ($800)
Offset 2680 ($a78): Class 7, private marker (SCSI2Pi end-of-data object)
```

The filename entry seems to have been written to the first block after the tape mark, not the second record. Do the first 3 records not have tape marks between them?

uweseimet, 2024-12-08:
OK, so you mean the filename should be at offset 356? Let's use offsets as a notation in this case, not block counts. This will avoid misunderstandings.

Pacjunk, 2024-12-08:
No the filename should be at 88. The HDR1 block.

The filename has been written incorrectly to 268 - leaving the HDR1 block at 88 still blank which is why it appears there are no files on the tape.

uweseimet, 2024-12-08:
OK, thank you. I will investigate this.

I think that without all these tools (s2pexec, s2ptool, s2psimh) it would be impossible to implement everything correctly. In theory you read the specification, implement based on it, and everything works. But in practice this is different. I will never understand why the PiSCSI project was not interested in such tools. s2pexec was already ready for a PR, and the lack of interest was a reason why I started SCSI2Pi.

When you created the PiSCSI tape ticket I labeled it with "major effort". I have not promised too much at that time, have I? ;-)

Pacjunk, 2024-12-08:
Often don't know until you start on something!  I think a better scsimon would help too! I had a look and can't make much of the output. I think its more about checking signal timings than tracing commands.

uweseimet, 2024-12-08:
Yes, scsimon is more about bus signals. But this is much too low level for what we are dealing with, because we are working on the SCSI command level. The logs for all signals would be really huge, and it would be impossible to correlate them to SCSI commands and their parameters.
And as already mentioned scsimon cannot capture all signals. In order to do that the Pi would have to be so fast enough to not only capture all signals, but also to save this information to the filesystem. But the Pi is not that fast, making a tool like scsimon useless.
If you are interested in bus signals you need an oscilloscope.

uweseimet, 2024-12-08:
For reference, if needed later, this is what your OS does. I verified all steps with my streamer and with s2p.

1. It reads blocks until it hits the first filemark at offset 264.
2. It spaces forwards until it is behind the last filemark, at offset 456.
3. At this point READ POSITION says that the current object location is 9.
4. It spaces backwards

Can it be that the last script/log also contained the mount operation? This is sometimes hard to tell, because the OS appears to use a set of building blocks for its command sequences. If the mount operation was included, please only add the copy operation next time. Re-starting s2p for that should be safe.

Pacjunk, 2024-12-09:
Build b1150f1.

Well that didn't write anything to the tape!  Before it wrote to the tape but in the wrong place. Now it doesn't write anything.

[copy.txt](https://github.com/user-attachments/files/18060113/copy.txt)
[copy script.txt](https://github.com/user-attachments/files/18060115/copy.script.txt)

uweseimet, 2024-12-09:
It is possible that the updated positioning code has changed the command sequence sent by your OS quite a lot. I will see soon.

uweseimet, 2024-12-09:
I compared the two copy scripts (from today and yesterday), and there is a difference that is not related to my changes.
```
-i 3:0 -c 00:00:00:00:00:00
-i 3:0 -c 12:00:00:00:24:00
-i 3:0 -c 1a:00:00:00:ff:00
-i 3:0 -c 1a:00:00:00:ff:00
-i 3:0 -c 15:00:00:00:0c:00 -d 00:00:10:08:00:00:00:00:00:00:00:00
-i 3:0 -c 1a:00:00:00:ff:00
-i 3:0 -c 34:00:00:00:00:00:00:00:00:00
```
In the script from yesterday your OS did not run READ POSITION (last line). It is running READ POSITION several times now, yesterday it did not do that. IMO this means that something has changed in the test scenario.
I mentioned that you could safely restart s2p before running the copy operation, because I assumed that yesterday you did not restart, but ran all operations in sequence. Can you confirm that it was different this time?

No need to re-run the test, at least not yet. First I need to know whether the scenario has changed.

Pacjunk, 2024-12-09:
I can't remember, but I don't think so. The errors are different, so maybe it is doing different commands trying to find the place to write.

uweseimet, 2024-12-09:
It can't be just like that, because it uses READ POSITION (34:00:...) right at the beginning before any command that can change the tape position. At that time nothing has really happened. This means that the reason why other commands are used is not related to the commands in the copy script.

Can you please run a test that also includes mounting the tape?

uweseimet, 2024-12-09:
What your OS was doing today is not that much different from yesterday, except for the final outcome. It's just the READ POSITION commands I stumble upon, because they were not there yesterday.

Pacjunk, 2024-12-09:
Mount & copy combined...

[mount&copy.txt](https://github.com/user-attachments/files/18061203/mount.copy.txt)
[mount&copy script.txt](https://github.com/user-attachments/files/18061204/mount.copy.script.txt)

uweseimet, 2024-12-09:
Regarding the write position, how do you know that it has to be position 88?
Even with my latest changes I wonder if they will really result in a different scenario. The OS sends a SPACE command to skip 32767 filemarks. Skipping filemarks implicitly also skips blocks. This command will only stop at end-of-data or end-of medium. The resulting position will be 456. Next, the OS spaces exactly a block back. This hits the last filemark, and the write position now is 452. The last log ends here, because the OS stops.
This looks like an append scenario to me. How does such a copy work when there are already data on a tape? Will it append to the existing data, or will it overwriting the data?

Pacjunk, 2024-12-10:
In case you find this interesting, I did the copy test under simh.

This is an initialised empty tape:
[1.tap.dmp](https://github.com/user-attachments/files/18076031/1.tap.dmp)

This is the tape after copying one file to it: (Note it goes at the HDR1 block, and also has an EOF1 block)
[2.tap.dmp](https://github.com/user-attachments/files/18076043/2.tap.dmp)

uweseimet, 2024-12-10:
These files are useful for testing locally, without your OS. Thank you.

Pacjunk, 2024-12-10:
Bingo, It works!  Copied a file to the tape and read it back successfully. Same label and filename/contents etc as the simh image I supplied earlier.

[copy.txt](https://github.com/user-attachments/files/18078099/copy.txt)
[copy script.txt](https://github.com/user-attachments/files/18078102/copy.script.txt)
[1.tap.dmp](https://github.com/user-attachments/files/18078105/1.tap.dmp)

I tried copying another small file and this also worked (two files on the tape). I then tried copying a larger file and this failed. The .tap file was still 64KB, so I assume the automatic appending to the file is not implemented yet!

Good to have a win!

uweseimet, 2024-12-10:
Great news! I am sure we will also solve the remaining issues. This is just a matter of time.

Appending has already been implemented, but it is not the default. See http://www.scsi2pi/en/properties.html for the respective "append" property.

Pacjunk, 2024-12-10:
I just copied about 1MB of files to a larger .tap file. Read back OK.  I will look into the properties.

uweseimet, 2024-12-10:
This would have to be something like "device.3.append=MAXIMUM_FILE_SIZE".

uweseimet, 2024-12-10:
By the way, the READ POSITION commands are gone ...

Pacjunk, 2024-12-10:
Yes, append works. I copied larger files and it extended.

As a bigger test, I mounted one of the simh tapes (the one I use for the backup/list test which contains about 50MB of data) on ID4 and copied this to a blank tape on ID3. Compared the two and they are identical.

Looking good.

uweseimet, 2024-12-10:
If you have other test scenarios (different than #111), please go ahead. If there is something else that does not work it may require a new ticket.
If there is nothing wrong with writing, we may close #112.

Pacjunk, 2024-12-10:
Both read and write drives were under s2p, if that was not clear.

Hard to tell at this stage, but writing looks good. I did get an error at one stage:

`[error] (ID:LUN 3:0) - Trailing record length 8192 at position 528 does not match leading length 8192`

but I had restarted s2p between copy commands, so it may have got confused. Didn't see that message again with later testing.

uweseimet, 2024-12-10:
Please add the commit ID you are currently using to this ticket.

A restart should not matter. Can you verify the error message with s2psimh? If there is something wrong with a file, s2psimh should also report that.

Pacjunk, 2024-12-10:
I have overwritten that file many times by now. It did not occur again. Build number is ab7b07b.

I'm trying to build a boot tape at the moment. I want to see if I can boot the machine off the tape. This will be a good test as it uses different drivers (ROM based instead of OS based).

uweseimet, 2024-12-10:
Sounds good!

Pacjunk, 2024-12-10:
Built the tape fine, but crashes when I try to boot off it. I will get you a log...

Pacjunk, 2024-12-10:
Short and sweet...

[boot.txt](https://github.com/user-attachments/files/18079083/boot.txt)
[boot script.txt](https://github.com/user-attachments/files/18079087/boot.script.txt)

Pacjunk, 2024-12-10:
Most of those messages came up after the system crashed...

BTW it boots off a hard drive image fine.

Pacjunk, 2024-12-10:
Weird - got the same crash message with s2p stopped. I think it must have left some line activated when it stopped as things only started working again when I turned the pi off. I will build a physical boot tape for comparison.

uweseimet, 2024-12-10:
In such a case, what can help is launching s2pexec and then run the "-r" command, which resets the bus.

Pacjunk, 2024-12-10:
It is so slow writing a physical tape!! s2p is much faster!

uweseimet, 2024-12-10:
You are using a Pi 3 now, don't you? I Pi 4 would probably even be faster. But a Pi 5 might actually be slower. The GPIO bus works differently with the Pi 5. As far as I can tell, with the Pi 5 you cannot transfer all 8 SCSI data bits in a single operation, because you need more than one GPI register for that. If this is really the case, would have a considerable negative impact.

uweseimet, 2024-12-12:
Any news? Note that I will most likely merge #107 into the issue_112 branch later today. I need to do that because #107 is essentially done, and maintaining two branches with a lot of changes in each of them is an inconvenient development overhead. The #107 changes should not affect the tape features, but I will add a note on the last commit ID before the merge when I am done, just in case.

Have you already tried USB mass storage drives with your workstation? After the merge this should work, and the drives should appear just like regular SCSI devices. SCSI2Pi will use the Linux SG driver to directly pass all commands to the connected drive. This is essentially the same as the pass through feature in virtualization tools like VirtualBox.

uweseimet, 2024-12-12:
As mentioned above, #107 has been merge into #112. The last commit before the merge was 71f3bbaa.
In case of major issues with the up to date branch due to the merge, testing #112 can temporarily be done on this commit.

Pacjunk, 2024-12-13:
Sorry, I've been away for a couple of nights, and tired tonight. I might have another play tomorrow.

uweseimet, 2024-12-13:
No problem, thank you for your note. There are other tickets I can work on ;-).

Pacjunk, 2024-12-16:
Build 6f568db

OK, I have swapped the workstation for another one (slightly faster and a later date on the SCSI ROM). This machine does not crash when attempting to boot the tape!  but you're going to love what it does!!!!!

[stb.txt](https://github.com/user-attachments/files/18145179/stb.txt)
[stb_script.txt](https://github.com/user-attachments/files/18145182/stb_script.txt)

uweseimet, 2024-12-16:
There is exactly one read and access, when this machine tries to read 512 bytes from an 80 byte block and fails. I don't think that the tape image is compatible with what the machine tries to do.

But before working on other topics, are there still any issues with writing to a tape that need to be resolved as part of #112?

Pacjunk, 2024-12-16:
I'm mostly happy with writing. Copying files onto the tape works. I just tried a backup directly to tape, which also seemed to work.
If I try to read or verify the backup. using a foreign mount, then it fails trying to read the 90 byte block as already discussed.
If I read the backup using a normal mount, then I can list the contents of the backup, but I get an error right at the end.

[end_read.txt](https://github.com/user-attachments/files/18148856/end_read.txt)
(This is the end of the log, the whole thing is too large, and irrelevant)

If I copy the backup file to disk and then read it, everything is OK - which shows that the backup was written correctly to tape. The error occurs during read.

So, overall, writing works - so I suppose this issue can be closed. The remaining issues are:

- Backup list/read/verify checks the tape label requesting a 90 byte block
- Booting a tape requests a 512 byte block
- Error at the end of a backup read (log above)

uweseimet, 2024-12-16:
OK, then let's close this ticket. In case of any future issues with writing we can create a new ticket any time.

I will create a separate ticket for "Error at the end of a backup read".

Regarding "Error at the end of a backup read", please have a look at the logfile, if you have not already done so. There is hardly anything happening, and the very first access to the tape results in an error. This error is IMO inevitable, because you cannot read 512 bytes when there is a 80 bytes block. Please run the respective script with s2pexec and your tape drive. I expect your tape drive to report the same error.

But before doing that, please update your issue_112 sources and re-compile, if you have not already done so.

Pacjunk, 2024-12-16:
I'm currently running 2155428

Pacjunk, 2024-12-16:
I think you can ignore that last log. Listing the backup on a real tape also gives the same error!  Strange...

uweseimet, 2024-12-16:
OK :). The log also does not indicate an error, but just the a filemark was found and that this was reported to the OS.

Pacjunk, 2024-12-16:
The usual way to list a backup tape is to mount/foreign - which works for the real tape. If not mounted foreign, you can still list but get the error. I'm beginning to wonder whether this DAT drive is 100% compatible, or just 99%?  I'm pretty sure the DEC official drives were just rebadged Sonys, but who knows what they did to the firmware.

Yes - I do look at the logs. Some things I understand, some not...

uweseimet, 2024-12-16:
This means that currently we have these 2 open items:

1. Booting a tape requests a 512 byte block, which according to the log is a justified error message
2. #111, with the "switch from s2p to streamer test" to be pending.

Please correct me if I am wrong.

uweseimet, 2024-12-16:
Regarding the tape drive, I doubt that there are firmware issues. When it is about reading more bytes than available, the specification is very clear: This is not possible, and a changed firmware that would do something else then it should is unlikely. Remember, your Sony drive also reports an error, just like s2p.

closed 2024-12-16 by uweseimet