uweseimet/scsi2pi #113: Create a tool that converts tape contents into SIMH .tap files

uweseimet, 2024-12-08:
In order to replace a real tape drive by s2p, a tool that converts all blocks and filemarks on a tape to the respective SIMH data is needed.
Such a tool, probably called "simhdump", shall be added.

uweseimet, 2024-12-10:
@pacjunk In the issue_113 branch there is an early version of this s2pdump extension. Restoring a tape is not yet supported, but converting a tape to a SIMH file works when running it against s2p like this:
```
s2ptool -s "-i 0 simh.tap" -c s2pdump -a "-i 0 -f /tmp/test.tap"
```
This will launch s2p with simh.tap as image file, and then run s2pdump against it. No Pi is required for this, because everything happens in memory. I will test with a real tape drive later, as good as this is possible with my drive.

uweseimet, 2024-12-11:
Merged into issue_112 branch.

closed 2024-12-11 by uweseimet

Pacjunk, 2024-12-16:
Must be doing something wrong...

```
pi@PiSCSI:~/scsi2pi/cpp/bin $ s2pdump -i 3 -f test.tap
----------------------------------------
Board (initiator) ID is 7
----------------------------------------
Checking SCSI target ID:LUN 3:0

INQUIRY product data:
Vendor:               'SONY    '
Product:              'SDT-5000        '
Revision:             '3.02'
Device Type:          Sequential Access
SCSI Level:           SCSI-2
Response Data Format: SCSI-2
Removable:            Yes
Error: Invalid device type for SCSI dump/restore, supported types are DIRECT ACCESS, CD-ROM/DVD/BD/DVD-RAM and OPTICAL MEMORY
Error: Can't get device information

```

uweseimet, 2024-12-16:
As far as I can tell this is not the s2pdump from the issue_112 branch.

Pacjunk, 2024-12-16:
I'm on the 112 branch, git pull says nothing to do. make s2pdump says nothing to do.

uweseimet, 2024-12-16:
This is what's in the issue_1122 branch:

```
        cerr << "Error: Invalid device type for SCSI dump/restore, supported types are DIRECT ACCESS,"
            << " CD-ROM/DVD/BD/DVD-RAM, OPTICAL MEMORY and SEQUENTIAL access" <<
```
I don't know what's wrong, but please re-build after a "make clean".

Pacjunk, 2024-12-16:
I did a make clean, also checked out develop, then back to issue_112. Rebuilt (with warnings), and same results.

uweseimet, 2024-12-16:
Please re-check your commit ID. The current one for issue_112 is 21554289.

Pacjunk, 2024-12-16:
No thats the path to the executables.

 How do I force it to re-download?

uweseimet, 2024-12-16:
Re-download? I'm afraid I'm not sure what you mean. In your branch with the compiled sources, run ./cpp/bin/s2dump. Or ./bin/s2pdump, depending on which hiearchy you are.

Pacjunk, 2024-12-16:
Yes, I do all that. Something is wrong. I have done another "git clone" and started again.

uweseimet, 2024-12-16:
The current issue_112 commit ID is 21554289.

Pacjunk, 2024-12-16:
No, same results. Too late now, I will try again tomorrow...

uweseimet, 2024-12-16:
OK. Note that because issue #112 is closed I will switch back to the issue_111 branch and will update it later today. Tomorrow please check out/update the issue_111 branch.

Pacjunk, 2024-12-17:
My commend seems to have disappeared, so I will write it again...

I tried dumping the tape, but it only gets to the first file mark (dump file has contents up to this point). Log attached:

[dump.txt](https://github.com/user-attachments/files/18162911/dump.txt)

uweseimet, 2024-12-17:
Is this with the current issue_111 sources? Please ensure that you are not using something older.

uweseimet, 2024-12-17:
OK. I don't know yet what's wrong, I am going to investigate.

uweseimet, 2024-12-17:
I guess this is a timeout issue, but this has not yet been logged. I have update issue_111 with better logging and a longer timeout. Please retry after updating and re-compiling.

Pacjunk, 2024-12-17:
That is a lot better, although I think there are still timing issues.

On Trace logging, the dump completed.
On info level logging, it died after about 5000 blocks with:
```
[error] [s2pdump] STATUS phase failed
Error: INFORMATION field is not valid
```

Is it possible to have some sort of progress indicator (e.g. a dot per block) rather than just no output until the end?

Also, if I do a Ctrl-C during the program nothing happens. Because there is no progress indication I don't know if its still going or hung. If I eject the tape, it still did not die.

Pacjunk, 2024-12-17:
Also the dump file is almost exactly double the size expected. When I list the tape in the OS, the files appear twice. The files look OK, but two copies!

uweseimet, 2024-12-17:
Two copies? s2pdump definitely only reads the tape once. Is it possible that you ran s2pdump twice and data were appended? But even this I cannot reproduce. Can you reproduce it?

Everything else I will most likely address as part of https://github.com/uweseimet/scsi2pi/issues/115 later.  "STATUS phase failed" indicates that there is a bus signal issue. I may have to improve the logging to see more. Is this reproducable?

In order to help with #111, as far as required, the current state of s2pdump should be fine, though. This is why I would like to deal with #111 again, which is related to the core functionality.

Pacjunk, 2024-12-17:
I think I now what might have happened. I will rebuild my "master" tape.

The error message only occurs on the "info" log level, not "trace". I assume the extra logging on trace slows it down just enough to get through.

Pacjunk, 2024-12-18:
I have successfully dumped and restored tapes using s2pdump. I will hopefully get back to some of the testing tomorrow.